prusa3d/PrusaSlicer #8084: "Relative extruder addressing  requires resetting the extruder position..." after upgrade from 2.4.0 to 2.4.1

kurt77, 2022-03-17:
### Version
2.4.1

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Anycubic Mega S

### Behavior
I receive the attached error as soon as i add any model on the build plate

[image]

rtyr, 2022-03-17:
Intended behavior, add G92 E0 command as instructed in the message (Printer settings -> Custom g-code)

[image]

From 2.4.1-beta3 release notes:

> In [2.4.1-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.1-beta1), we added a check that G92 E0 is present in layer change G-Code when relative extruder addressing is used. This is needed to prevent error accumulation in firmware. From this version, the check is only applied for Marlin firmware, because as far as we know, other firmware flavors do not suffer from this issue. https://github.com/prusa3d/PrusaSlicer/issues/7846
> 
> Related to the previous improvement: our G-Code preview works by mimicking the Marlin firmware, therefore if the G92 E0 was missing and the error accumulation became visible, it would also be visible in the G-Code preview. Because only Marlin firmware suffers from this issue in reality and there is an explicit check for the G92 E0 when Marlin is selected, we have now modified the G-Code Preview so the loss of precision problem is not visualized. https://github.com/prusa3d/PrusaSlicer/issues/4881, https://github.com/prusa3d/PrusaSlicer/issues/5073, https://github.com/prusa3d/PrusaSlicer/issues/6336

Closing.

closed 2022-03-17 by rtyr

Adorfer, 2025-01-28:
it's still in OrcaSlicer 2.2.0 (stable release) in the default of Anycubic/Vyper, when adding the printer the first time to the setup from the provided templates. 
Expected behavior "G92 E0": should be already in the available templates. (and not required to be added by the user manually)

tekchip, 2025-02-02:
Can confirm this is still happening. Orca 2.2.0, Windows.

rtyr, 2025-02-02:
This is PrusaSlicer repository.

tekchip, 2025-02-02:
> This is PrusaSlicer repository.

You're right. My bad. Oops.