prusa3d/PrusaSlicer #8086: Wipe tower extruding blobs causing hot end to hit it

ajq11, 2022-03-18:
### Version
2.4.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 (build 22000), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Creality CR-6 SE w/ 3D Chameleon

### Behavior
* When it extrudes the colors, before the wipe tower and on the wipe tower it extrudes a whole bunch of filament which makes it not extrude smoothly. So, the hot end, ends up colliding with these bumps on following layers.
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]

lukasmatena, 2022-03-18:
These purge lines can be turned off in Print Settings, but they should not be overextruded and they should also be printed at the edge of the bed so the hot end should not travel over them. Please, post a 3MF demonstrating the issue. Thanks.

lukasmatena, 2022-03-18:
> Please, post a 3MF demonstrating the issue. Thanks.

ajq11, 2022-03-18:
[CHEPCalibrationCube.zip](https://github.com/prusa3d/PrusaSlicer/files/8306109/CHEPCalibrationCube.zip)
It doesn't just let me upload the 3mf file type, so I zipped it. Hope that works.

ajq11, 2022-03-18:
[image]
Here is another photo of the actual wipe tower having the issue.

Bendermk6, 2022-03-26:
[image]

You have to edit the extrusion setting in your tool change G-code.
`G0 E90 F2000 ; <<<<<-----------  EDIT THIS DISTANCE TO FINE TUNE THE FILAMENT LOADING TO STOP BLOBBING OR UNDEREXTRUSION`

ajq11, 2022-03-26:
I have adjusted that so it doesn't blob initially, but after it starts extruding it spits out way more filament than the speed it is moving, causing the over extrusion bumps on the wipe tower (at the beginning), then settles down and prints normally.

rtyr, 2022-03-26:
I think you have most likely just misconfigured "Single extruder MM setup" (printer settings).

Bendermk6, 2022-03-26:
> I think you have most likely just misconfigured "Single extruder MM setup" (printer settings).

Yes, i think the same. Because with a MMU2s you got different positions in it. maybe you can start with decreasing the extra loading distance from -2 -> -10
[image]

rtyr, 2022-05-03:
No response. The amount of loaded filament is based on the "single extruder MM setup" values, so it is most likely just misconfigured. Closing.

closed 2022-05-03 by rtyr