prusa3d/PrusaSlicer #8091: Design bug: Yes/No response to save dialogue

merchako, 2022-03-18:
### Description of the bug

#### Background
When a user closes PrusaSlicer with an unsaved file open, the user is prompted whether to save the file or discard it.

#### Problem
The dialog uses the generic, hard-to-scan responses "yes" and "no", so 

[image]

#### Solution
From [Jakob Nielsen of NN group](https://www.nngroup.com/articles/confirmation-dialog/):

 > Instead of Yes/No answers, provide response options that summarize what will happen for each possible response. For example, in the case of file deletion, use buttons labeled Delete file and Keep file.

Instead of "yes", "cancel", "no", let the buttons say "Cancel",  "Discard", and "Save".

### Project file & How to reproduce

When a user closes PrusaSlicer with an unsaved file open, the user is asked whether to save the file or discard it. You can reproduce this with any unsaved file, so I have not included a project file.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+arm64

### Operating system

 macOS Version 12.2.1 (Build 21D62)

### Printer model

Ender 3 Pro

HaasNL2, 2022-03-18:
Totally agree with this.

FidelCapo, 2022-04-11:
Thank you for reporting. This bug has been fixed and will be part of the next release.
Closing.

closed 2022-04-11 by FidelCapo