prusa3d/PrusaSlicer #8095: MMU2S - Adding Text to Printed Parts

Noggin01, 2022-03-19:
**Is your feature request related to a problem? Please describe.**
Using a permanent marker to write a serial number on printed parts results in the ink seeping into the seams between extruded lines. This does not look appealing.

**Describe the solution you'd like**
I would like an option to add text to a part before slicing. There is already an MMU painting feature in PrusaSlicer, but it doesn't support adding text to the print. I'd like to be able to add text (or better, a serial number) to a part before or during slicing.

**Describe how it would work**

1. Select paint mode
2. Draw a box on the surface of a part to have text added
3. Enter text into a text editor and have it render the text on the 3D model
4. Bonus - Allow for text size to be increased automatically if the font is too small for the selected nozzle
5. Super awesome bonus - Allow for serial numbers to be added. The text box could ask for a prefix (such as "SN: ") and then a format for the serial number (4 digits max, include/exclude leading 0's, utilize a date, etc), and then how many parts to generate
6. Hit the "Slice now" button

In the case of serial numbers, this would require generation of multiple parts. For instance, I might enter "17" as a starting serial number and have it generate 5 parts. This would result in 5 g-code files being generated, each one including the "SN: 0017" text in the file name.

**Describe alternatives you've considered**
Aside from obvious things like using stickers or a different type of pen/marker, I've considered using SCAD to generate serial number labels. I think I can place my object into PrusaSlicer, add a rectangular mesh and use a boolean operation to cut a section out of my object, then add an SCAD generated 3D model of a serial number label. SCAD is great for generating things by script, so I could easy generate a lot of serial numbers in this fashion, but I'm not sure I can make them be separate bodies. I think the text body would merge with the surrounding plastic, and I'd end up with a simple rectangle.

plampix, 2022-03-29:
They're already working on text objects, and it works quite nicely. See for example this Benchy with a text object added as a negative volume. You'll be able to mirror the text on all of the axes, and edit text afterwards.

[image]
[image]

rtyr, 2022-05-17:
Closing as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2296. This feature is being developed.

closed 2022-05-17 by rtyr