prusa3d/PrusaSlicer #8096: [Feature request] Tool to split models and bridge the split seamlessly to stretch sections of models.

KriLL3, 2022-03-20:
I've repeatedly encountered an issue where I've made a complex design and some dimension isn't quite right, but since the part isn't simple enough, simply scaling the part doesn't fix the problem.

A tool that works similar to split but basically splits the part, moves one piece away from the other, and bridges the gap with new geometry that has the same profile as the cut would be very useful and save me a lot of time when this happens, alternatively could define 2 planes (to make 3 parts) and stretch the center part to achieve the same effect.

It is possible to achieve a similar result by splitting a part into 3 pieces, scaling the middle piece and arranging them all manually, but it would be a lot less hassle if it was a proper feature.

dartrax, 2022-03-23:
Duplicate of #7563

KriLL3, 2022-03-23:
> Duplicate of #7563

Yeah, sorry, I searched for an already existing suggestion like this but didn't find that one, that title isn't the greatest heh.

rtyr, 2022-03-23:
Duplicate. Closing.

closed 2022-03-23 by rtyr