prusa3d/PrusaSlicer #8097: multi part stl import crash

Vance1122, 2022-03-20:
### Description of the bug

version 2.4.1
When I go to open an stl that has multiple non connected parts in it, Prusa Slicer just shuts down
[Ring Of Fire.zip](https://github.com/prusa3d/PrusaSlicer/files/8312208/Ring.Of.Fire.zip)
.

### Project file & How to reproduce

[Ring Of Fire v49_Ring Of Fire v49_The Ring Of Fire_1_The Ring Of Fire Base_.zip](https://github.com/prusa3d/PrusaSlicer/files/8312221/Ring.Of.Fire.v49_Ring.Of.Fire.v49_The.Ring.Of.Fire_1_The.Ring.Of.Fire.Base_.zip)
This file is ok, but it has the profile I am using. The file that is causing problems is in the description.
You add the file to the plate and prusa slicer puts up the widow telling me that the 3mf file has multiple parts in it and before I can select how I want it to handle it, prusa slicer shuts down so I never get to the point where it is on the build plate ready to be sliced.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1+Linux-x64-gtk3

### Operating system

Mint Linux 5.4.0-104- generic x86 64

### Printer model

Prusa MK3S

jusii, 2022-03-21:
Can confirm with 3mf multipart file and 2.4.1+Linux-x64-gtk3 loaded 3mf exported with 3D Builder and instant crash when opening.

[2022-03-21 07:31:39.614834] [0x00007f928e7128c0] [trace]   Initializing StaticPrintConfigs
free(): invalid pointer
Aborted (core dumped
[ear_R.zip](https://github.com/prusa3d/PrusaSlicer/files/8313306/ear_R.zip)
)

lukasmatena, 2022-03-21:
Thanks. This seems to be a duplicate of #8078. It is being investigated.

lukasmatena, 2022-04-05:
This should be fixed in the just released 2.4.2-rc1. Can you please confirm? Thanks.

jusii, 2022-04-06:
Hi, seems to work with my sample file. Thanks!

Vance1122, 2022-04-06:
I got the closing the app without saving first worked out.

closed 2022-04-06 by lukasmatena