prusa3d/PrusaSlicer #8098: Pause Print stopped working in this release?

becktuning, 2022-03-21:
### Description of the bug

when adding a pause in slic3r 2.4.1 the printer displays the pause text on the screen but continues to print and not pause. I tried this on both the MK3s and prusa MINI+ 

### Project file & How to reproduce

add a pause to the print at any layer height.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Mac OS 12.2.1

### Printer model

MK3s and Prusa Mini+

rtyr, 2022-03-22:
PrusaSlicer only emits M601 command (unless specified differently in printer settings), it still work the same way in 2.4.1. Are you printing directly from USB/SD card? It is possible the M601 command is rejected by Octoprint for example.

Anyway, please provide your 3MF project file so we can check.

becktuning, 2022-03-25:
I am running octoprint I rolled back to the previous version of prusaslicer and it works correctly again.

rtyr, 2022-03-25:
Was it with the Prusa MINI? I think this issue is same as https://github.com/prusa3d/PrusaSlicer/issues/4573.

The thing is, that Prusa MINI apparently doesn't yet support pausing over serial/Octoprint (M601 command). It only works when printing from flash drive. 

Here is the firmware issue related to this problem:
https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/283

I am going to close this issue, because it's not related to PrusaSlicer, but I would like to check your problematic 3MF project file just to be sure.

becktuning, 2022-03-27:
the issue was on the MK3s but I did try it on my prusa mini+ also and same issue

rtyr, 2022-03-27:
OK. Please provide some of the problematic gcode or 3mf, so I can confirm that the issue is not on the PrusaSlicer side.

rtyr, 2022-04-11:
No response. Likely not PrusaSlicer issue. Closing.

closed 2022-04-11 by rtyr