prusa3d/PrusaSlicer #8099: Stop storing printer configuration in the project file, or prompt if you want to use the old settings when you open it

traderhut, 2022-03-21:
Not sure if this is a feature, requesting a removal of a 'feature' or if this is a bug.. but

Consider this situation... you have 3,423 projects that you have worked on.  You have save each of these with the current printer, let's call it 'Tool Changer'

Now, you realize that there is a much better setting in the Tool changer printer, so you want to update that.

Now, when you open Project 1, you get "Tool Changer (modified)" as your settings.  You have to revert those changes (and OMG that is a bitch to do from what I can tell.. Involving reverting it, exiting the application, saying "Dont' save changes" and re-entering the application... Ugh.

Now, to fix the other files, just repeat that 3,422 more times and you have fixed your printer!  Done, in only 4 MONTHS to change ONE setting on your printer!

If you MUST save the printer specific settings in the project (and it seems like a non-crazy idea.),   Could you prompt me when I open the project:  "Your printer/other settings" have changed since you last opened this file.  Would you like to use the new settings.

(which, will be 'Yes' 99.97% of the time btw)

Even better, you could have an option to save last successful print's settings or something if you want to be able to revert to what you had for your printer configuration 2 years ago when you last printed this..

traderhut, 2022-03-21:
A more common situation:  Let's say you have a multi color printer, and you load up Red in Extruder 0, and Green in Extruder 1.
You want to have the parts that are printing with each show up in that color, so you set the colors to match on your printer configuration.

Now, you edit and save a few files.

You load new Green and Yellow instead, and update your printer to match

Every time I go pull up an old file, it shows it with the wrong colors puts (modified) on the settings... And the UI is so convoluted on the savings of those setting / resetting them that it is hard to figure out how to revert the 'changes' you just made - which, of course, you didn't really make any changes, you just opened a project.

The first time, and again every time I've reinstalled this software, I've assumed that there was no way to create a new Preset - I had to hit the 'Save' and have it prompt me.   99% of the time, when I save, I just want to save, the other 1% of the time, I'd never guess that it was going to prompt me if I hit save.  This entire UI scheme needs redesigned to be intuitive.   It being so hard to figure out what I'm doing is the #1 reason that I go back to using Repitier Host for my slicing...  As this UI drives me crazy.

I would want to see:   Save, Save As, Discard changes (or Reload from default whatever you want to call it), New preset, and delete on each page instead.

I still don't know why my printer is called 'Tool Changer + Tool Changer'   

I would NEVER have guess that to revert my changes, I would select 'Tool Changer (modified)' from the drop down, I want the UNMODIFIED version, not the modified one, so I only selected that one by accident to get it to revert (it prompts finally)

Like I said, this is the top reason that I stop using this software... Just to cra-cra for me as a UI...

traderhut, 2022-03-21:
Another example:   I go in, my settings are modified (because I opened a project), I have Tool Change + Tool Changer selected as my printer settings  drop down.  I select user pre-set of 'Tool Changer' which switches it away and asks me if I want to dump all my changes... YES!  I do that, and it then forgets that I'm using a printer, so I re-select that, go back to the Plater page, don't have the option to export directly to the printer.  Save, exit, go back in, and reload the project.  Now I have the ability to print directly to the printer and my settings got updated.

I have to do this every time I change a setting to get it to pick it up in a project other than the one I'm in.

rtyr, 2022-03-22:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/6377, which I reopened. 

Project file should contain all settings by definition. You can choose whether to import 3MF with or without settings, so it should not be a problem. (Add... button or Add. option will import 3MF without settings, drag and drop will ask). The only scenario which is likely not covered is opening 3MF by file extension association.

closed 2026-09-29 by prusaslicer-bot[bot]