prusa3d/PrusaSlicer #8103: Supports above model

OivindAlvestad, 2022-03-22:
### Description of the bug

I tried using support for my model of a ring, and chose "support everywhere", and for some reason, the slicer creates support above the model.

### Project file & How to reproduce

[SupportAboveModel.zip](https://github.com/prusa3d/PrusaSlicer/files/8322638/SupportAboveModel.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0

### Operating system

macOS Monterey 12.2.1

### Printer model

Prusa Mini

rtyr, 2022-03-22:
Screenshot
[image]

Thanks.
You can change support style to "Snug" to prevent this.

OivindAlvestad, 2022-03-23:
Thanks:) I just used the support enforcers to get around the problem, so it was no big deal, I just thought it was weird that it could even make support above the model. But I've never tried the snug support, learned something new today:)

closed 2022-03-23 by OivindAlvestad