prusa3d/PrusaSlicer #8106: SD Card Ejection / remove device, seems to not work correct in windows 11, occasionally corrupts SD cards

acekrystal, 2022-03-23:
### Description of the bug

**Silcer Version :** 
```
Version: 2.3.0+win64
Build: PrusaSlicer-2.3.0+win64-202101111315
Operating System: Windows
System Architecture: 64 bit
```

**OS**
Prusa Info : 

> Windows Version: Windows 10 (build 22557), 64-bit edition

Actual windows System info : 

> Edition : Windows 11 Home
> Version : Dev
> OS build : 22557.1
> Experience : Windows Feature Experience Pack 1000.22557.1.0

**Behavior**
After slicing I export the G.code to the SD-card using the special SD-card button in the slicer. After it saved to the SD-card you can press an "eject" button to safely remove the SD-card. Though it does seem to do "something" in windows, it does not do the same thing as ejecting the SD-card using windows explorer. And about 15% of the time I now get an corrupted SD-card that shows up empty in the printers (MK3(S+)) and when putting back in the PC it now wants to format the SD-card.

The difference between Windows ejection and slicer ejection : 
**Windows eject:** 
- LED light on SD-card reader is turning OFF.
- SD-card storage disk is fully disappearing from explorer "This Computer"- disk overview.
- On right bottom menu bar removable device icon, right mouse does not seem to show any disks attached

**Prusa eject :** 
- Led stay's ON on SD-card reader
- SD-card storage disk is still visible in explorer "This Computer"- disk overview, but I can't open it or see how much disk space is left.
- On right bottom menu bar removable device icon, right mouse shows a device "..." but I can't do anything with it.

**Windows (11) Eject changed?**
I do need to note that in the past (month ago?) Windows Eject did do the same thing as in slicer Prusa eject did, resulting in also corrupt SD-cards with the windows eject. Currently I noticed a difference between the Windows way of ejecting and the Prusa way of ejecting, so it seems Windows 11 changed something and maybe prusa needs to take a look at what they changed for ejecting removable disks. 

I have not confirmed yet that the seemingly new windows way of ejecting also solves the 15% corrupt SD-cards, but time will tell and the fact that the led light is literally turning off gives me good hopes.

> 
> Extra info : 
> - Tested with multiple SD-cards, and multiple different brand USB-SD-card readers. all have same behavior.

### Project file & How to reproduce

1. Open any model in slicer (on windows 11)
2. Insert SD-card in pc (windows 11)
3. slice your file
4. Save slice file (Gcode) to SD-card using the buttons inside slicer software
5. Use "eject" button inside slicer to eject your SD-Card "safely"
6. Insert in Prusa printer (MK3(S+))
7. You will see the printer only showing ".." and no other files.
8. Placing back in windows PC, PC will now ask to format the SD-card, else the SD-card is not usable.

In my experience about 85% of the times I have no problems, but 15% of the cases SD-card becomes corrupt, and I lose all sliced files on the SD-card :(

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version: 2.3.0+win64 Build: PrusaSlicer-2.3.0+win64-202101111315 Operating System: Windows System Architecture: 64 bit Windows Version: Windows 10 (build 22557), 64-bit edition

### Operating system

Windows 11

### Printer model

Prusa MK3(S+)

lukasmatena, 2022-03-24:
Thanks for the report. I was not able to reproduce the issue. Can you please try with the recently released PrusaSlicer 2.4.1? I do not recall anyone would fix a bug like this, but it would be good to investigate on the latest version.

There were other issues reported with ejecting SD cards and it often ended up being caused by a cheap and substandard SD cards. If you also had these issues when ejecting normally (outside PS), it is quite possible it is the case. Do you have this problem with only one SD card or any?

It may be somehow connected to #4583, but I have no clue how exactly.

OniriCorpe, 2022-07-30:
I have the same problem with Windows10 and the USB key provided with the prusa mini plus.

closed 2026-09-29 by prusaslicer-bot[bot]