prusa3d/PrusaSlicer #8107: "External Perimeters First" enabled seems to ignore seam setting

jurassic73, 2022-03-23:
### Description of the bug

Pics here:
https://forum.prusaprinters.org/forum/prusaslicer/does-external-perimeters-first-somewhat-ignore-your-seam-setting/

With the model, I print with these two settings and my seams look good:
External Perimeters First: disabled
Seam Position: nearest

If I change to this, my seam position prints like it's random and it's very visible:
External Perimeters First: enabled
Seam Position: nearest

### Project file & How to reproduce

Is this export public or private to Prusa?  I have designs I print to sell and don't want to post if this is public.

With the model, I print with these two settings and my seams look good:
External Perimeters First: disabled
Seam Position: nearest

If I change to this, my seam position prints like it's random:
External Perimeters First: enabled
Seam Position: nearest

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.1+win64

### Operating system

Win 10 64 bit

### Printer model

MK3s

lukasmatena, 2022-03-24:
> Is this export public or private to Prusa?

It is public. Are you able to reproduce the issue on a free model so you can send a 3MF containing that? It is difficult to discuss issues like this if we don't all look on the same thing. If you are not, I will give you an email address to send the model privately.

bubnikv, 2022-06-29:
The issue seems to be fixed with the seam refactoring in the upcoming PrusaSlicer 2.5.0-alpha2. Would you please confirm that your issue is fixed?

closed 2026-09-29 by prusaslicer-bot[bot]