prusa3d/PrusaSlicer #8109: Macros not processed correctly

LittleHobbyShop, 2022-03-24:
### Description of the bug

Macros are not calculating correctly (or per the documentation)
Documentation is found [here](https://help.prusa3d.com/en/article/macros_1775)

The temp tower example given being:
`M104 S{if layer_z < 10}265{elsif layer_z > 45}240{else}265+(240-265)(layer_z-10.0)/(45-10){endif}`

This results in a gcode output of this:
`M104 S265+(240-265)(layer_z-10.0)/(45-10)`

Any calculations after the {} operator is ignored and simply output as a string

### Project file & How to reproduce

Use the example mentioned above

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows

### Printer model

Custom

lukasmatena, 2022-03-24:
> `M104 S{if layer_z < 10}265{elsif layer_z > 45}240{else}265+(240-265)(layer_z-10.0)/(45-10){endif}`
This results in a gcode output of this: `M104 S265+(240-265)(layer_z-10.0)/(45-10)`

Of course it does. That's what is written there. According to the documentation, expressions are evaluated when they are in curly braces. Yours is not. You want `M104 S{if layer_z < 10}265{elsif layer_z > 45}240{else}{265+(240-265)*(layer_z-10.0)/(45-10)}{endif}`.

> In addition, using the same example but with the ternary operator gives the following error:

Because it is incorrectly parenthesized. The documentation says that the whole ternary expression shall be in round parentheses. Yours is not. `M104 S{((layer_z < 10) ? 265 : ((layer_z > 45) ? 240 : 265+(240-265)*(layer_z-10.0)/(45-10)))}`

Next time please attach project file(s) when asked for it, regardless of how easy you think reading of the text is. Thanks.

Can we close this?

LittleHobbyShop, 2022-03-24:
@lukasmatena These are not my examples, they're directly copied from the knowledge base page linked and are given as examples.

In this case then the knowledge examples are incorrect and need to be updated. There also seems to be a minor issue with some wrapping of text as code format <>

[image]

lukasmatena, 2022-03-24:
@LittleHobbyShop Thanks. Only now I understood that you are quoting verbatim examples from the manual. I though those are your constructs and you refer to a manual that you do not follow correctly.

In that case, you are right that there is a problem. And as far as I can tell, the problem is not PrusaSlicer, it is really the manual. I believe that the example from the documentation never worked in any PrusaSlicer version. I tried in an ancient Slic3rPE 1.39 to make sure and it behaves the same.

I have fixed both the examples and the incorrect code highlight. You can check. Thank you for reporting the problem.

Can I close this now? :-)

LittleHobbyShop, 2022-03-25:
Perfect! Thanks, yes I think it can be closed.

closed 2022-03-25 by lukasmatena