prusa3d/PrusaSlicer #8112: Build doesn't work on Windows, Can't find boost when building deps

traderhut, 2022-03-24:
### Description of the bug

Running the command:

build_win -d=..\PrusaSlicer-deps -r=console

and I get

CMake Error at C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: Boost_INCLUDE_DIR system filesystem thread
  log locale regex chrono atomic date_time iostreams) (Required is at least
  version "1.66.0")

Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Professional/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.20/Modules/FindBoost.cmake:2344 (find_package_handle_standard_args)
  CMakeLists.txt:362 (find_package)

I get the same error if I go into the folder and do the crazy build command to build the deps (with a full path off the root down to the folder that I'm in)

I'm using Visual Studio 2019 Professional, starting it from the VS Command window, and I have the source in C:\src\3D\PrusaSlicer\PrusaSlicer

(I had to do two levels, as this thing goes off and creates a folder outside of the git directory (..\PrusaSlicer-deps ), so clearly, I don't want that in my src folder, so created another folder)

Running on Windows 10

### Project file & How to reproduce

None

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

lastest master branch as of 3/24/2022 5:15 PM

### Operating system

Windows 10

### Printer model

Tool Changer

traderhut, 2022-03-24:
I guessed that maybe I could open the ALL_BUILD file and use it to build the deps, it appears that it works (it is a VS 2017 file btw, not 2019 like everything else)... That build is in progress...  I'll then have to go to the main project and try to build that.

The original issue that the build instructions don't work is still valid.

traderhut, 2022-03-25:
I (mostly) got the deps folder to build - but got an error 'cmd exited with return code of 1' from one of them (but I don't see any error, so maybe it worked)

I built it in 2019...  I couldn't get the app to build after building...    I don't see a Prusaslicer.sln file anywhere.. so not sure how to build it from within Visual Studio

tamasmeszaros, 2022-03-30:
@jschuh Any ideas?

foreachthing, 2022-03-31:
For some reason, I only got this `build_win.bat` working only once. It built everything and never did a rebuild.

I still rely on my scripts. First, to pull and build deps:

I assume you're in `c:\dev\`.
Copy and create `c:\dev\PrusaSlicer_noob_build.bat` (or whatever name you can think of) with this content:

```
@echo off

REM ECHO I'm going to CLONE the PrusaSlicer repository - ready?
REM ECHO Best, if you're curently in c:\dev (or similar short path)
REM ECHO Also make sure you got at least 15 GB of free disk space available
REM ECHO.
REM ECHO Press CTRL+BREAK if you want to cancel.
REM PAUSE
REM ECHO Are you sure??? Shall I continue?
REM PAUSE

git clone https://github.com/prusa3d/PrusaSlicer.git --recursive

cd PrusaSlicer
SET "thisPATH=%CD%"
echo Building in %thisPATH%

cd deps
mkdir build
cd build

cmake .. -G "Visual Studio 16 2019" -DDESTDIR="%thisPATH%"

REM This will take a loooong time! Go get some coffee...
msbuild /m ALL_BUILD.vcxproj

cd "%thisPATH%"
mkdir build
cd build
cmake .. -G "Visual Studio 16 2019" -DCMAKE_PREFIX_PATH="%thisPATH%\usr\local" -Wno-dev -Wno-error=deprecated

msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal

```

And then the script (`c:\dev\PrusaSlicer\PULL_AND_RUN_BUILD_PrusaSlicer.ps1`) to be run in "Developer PowerShell for VS2019" to rebuild (and copy the relevant files to `c:\dev\PrusaSlicer_Release`):

```
# PS Script to PULL and REBUILD PrusaSlicer from Source

$gitstat=git pull --verbose
$gitcommit=git rev-parse HEAD
$gitdescribe=git describe --tag

$done=0

$StartDate=Get-Date

function RebuildPrusaSlicer {
	cd build	
	msbuild -ds -nologo PrusaSlicer.sln -t:rebuild -m -p:Configuration=Release -verbosity:normal
	cd ..
}

function SyncRelease {
	Write-Output ""
	Write-Output "====== Mirror to local machine ======"
	robocopy "c:\dev\PrusaSlicer\build\src\Release" $("c:\dev\PrusaSlicer_Release") /MIR /xf *.exp *.lib *.pdb /R:600 /W:30
}

If (($gitstat -join '') -eq ('Already up to date.')) { 
	
	Write-Output ""
	Write-Output "### Already up to date."
	Write-Output ""
		
	$confirmation = Read-Host "Do you want to copy the current build anyway (y/n)? Or do you want to force rebuild and sync (r)? Or just sync (s)."
	if ($confirmation -eq 'y') {
		SyncRelease
		$done=1
	}	
	Elseif ($confirmation -eq 'r') {
		RebuildPrusaSlicer	
		SyncRelease
		$done=1
	}
	Elseif ($confirmation -eq 's') {
		SyncRelease
		$done=1
	}
	Elseif ($confirmation -eq 'n') {
		$done=1
	} else {
		$done=0
	}

} Else { 

	Write-Output ""
	Write-Output "====== Rebuilding now. This will take some time! ======"
	
	Start-Sleep -s 2

	RebuildPrusaSlicer
	
	SyncRelease
	
	
	$EndDate=Get-Date
	$DURATION = New-TimeSpan -Start $StartDate -End $EndDate
	Write-Output "$('Total Build Time: {0:hh\:mm\:ss\:fff}' -f $DURATION)"
	
	$done=1

}

if ($done -eq 1){
	Write-Output ""
	Write-Output $("Commit now at : " + $gitcommit.SubString(0,7) + " --> " + $gitdescribe)
	Write-Output ""
} else {
	Write-Output ""
	Write-Output $("Commit now at : " + $gitcommit.SubString(0,7))
	Write-Output $(git describe --tag)
	Write-Output "Build and sync aborted"	
	Write-Output ""
}
```

jschuh, 2022-03-31:
@traderhut Seems like one or more of the downloads during the deps build failed. Unfortunately, the deps build process is a bit quirky, and can fail in unsalvageable ways (in my experience mainly due to download issues). If it fails I just rerun it with something like this to force a clean rebuild (which also redownloads everything):

```
build_win -d=..\PrusaSlicer-deps -s deps
```

@foreachthing I expect your problem was just that you weren't rebuilding the deps after they changed (since I noticed that your script always cleans and rebuilds the full deps). Sometimes the deps will go for weeks without changing; sometimes they'll change multiple times in a week. Either way, building the app will fail if you don't rebuild the deps after they change.

I automated the handling of deps changes in the scripts I use to manage my own worktrees. Take a look at [the relevant section in this wrapper script](https://gist.github.com/jschuh/275de2eab6944c36453513d53ed0000c#file-ps_update_worktree-bat-L77), which tracks the git revision history for the deps directory and rebuilds the deps if they've changed since the last successful build. First it tries a dirty rebuild (which sometimes works) and then falls back to a clean rebuild (which unfortunately is the norm).

jschuh, 2022-03-31:
Thinking about it, I could author a PR that incorporates the deps tracking and retry code from my own build scripts. I hadn't done that because currently nothing in the build system depends on git, but I suppose I can just ignore that step if git isn't present or the build isn't being run in a git checkout.

traderhut, 2022-03-31:
> For some reason, I only got this `build_win.bat` working only once. It built everything and never did a rebuild.
>
> I still rely on my scripts. First, to pull and build deps:
>
> I assume you're in [...]

I happened to have an (almost) empty C:\Dev, one unrelated folder.  (I normally use \src, but used \dev for this test)

I copied everything in the first batch file into a batch file named .\prusaSlicer_noob_build.bat

and ran it from the Powershell for VS 2019 window.

It did a lot, ending after a bit (not a loooong time, but my machine is crazy fast with SSDs that do 3GB/sec I/O and a 8700K overclocked CPU - under 5 min) with 12 warnings and 1 error:
       "C:\dev\PrusaSlicer\build\PrusaSlicer.sln" (rebuild target) (1) ->
       "C:\dev\PrusaSlicer\build\ALL_BUILD.vcxproj.metaproj" (Rebuild target) (4) ->
       "C:\dev\PrusaSlicer\build\src\libslic3r\libslic3r_cgal.vcxproj.metaproj" (Rebuild target) (23) ->
       "C:\dev\PrusaSlicer\build\src\libslic3r\libslic3r_cgal.vcxproj" (Rebuild target) (44) ->
       (ClCompile target) ->
         C:\dev\PrusaSlicer\src\libslic3r\ExPolygon.hpp(175,29): error C2065: 'p1': undeclared identifier (compiling source file C:\dev\PrusaSlicer\src\libslic3r\MeshBoolean.cpp) [C:\dev\PrusaSlicer\build\src\libslic3r\libslic3r_cgal.vcxproj]

This looks like a bad check-in?

Not tried the 2nd script yet.

traderhut, 2022-03-31:
Running 2nd script took 15 seconds..., It prompted me if I wanted to rebuild/sync, I entered 'r' to just rebuild (already should be sync'ed), and it ended with the same error about p1 not being defined.

traderhut, 2022-03-31:
PS C:\dev\PrusaSlicer> git status
On branch master
Your branch is up to date with 'origin/master'.

traderhut, 2022-03-31:
Yeah, it looks like a bad check-in, I did a git checkout 8b73d8b65aafad59058fa0e6cc0af1f7721cd672

And it seems to be building.. (Yesterday's last checking)
Has been building for the last 2minutes anyway..

traderhut, 2022-03-31:
What is the stable /release branch?

foreachthing, 2022-04-01:
> automated the handling of deps changes in the scripts I use to manage my own worktrees. Take a look at [the relevant section in this wrapper [...]

I'm no programmer 😉 so please go easy on me.

I tried your wrapper script after a fresh clone of PrusaSlicer and it failed:

```
C:\dev\PrusaSlicer>ps_update_worktree.bat
********************************************************************************
Updating and building master for remote origin/master
********************************************************************************
** Updating Repo: master
fatal: 'upstream' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
Maybe you know why?!

> First it tries a dirty rebuild (which sometimes works) and then falls back to a clean rebuild (which unfortunately is the norm).

Yeah, I have to do the same with my scripts. If build fails; start over!

As of now, your build_win.bat seams to work. Maybe the last fixes helped, since I got VS2019 AND VS2022 installed.
Will try a rebuild of the app as soon as our slow VM (~24 minutes) finishes...

iovispot, 2022-04-01:
Would be helpful if there was already a .sln file instead of leaving us to deal with the cursed CMake chicanery, since the point should be avoiding using it in the first place.

jschuh, 2022-04-01:
> I tried your wrapper script after a fresh clone of PrusaSlicer and it failed:

Oh, I wasn't suggesting you use my script. It relies on my worktree and directory structure, and will fail outside of that. I was just pointing to the part of the script that demonstrates how to automatically detect a deps change and rebuild only when necessary. That way you don't have to burn all the cycles and bandwidth on unnecessary rebuilds.

I have scripts that build all my PRs daily, and they almost never do clean rebuilds. The only exceptions are that deps usually need a clean rebuild after a deps change (as mentioned) and occasionally an MSVS minor version update will require a rebuild (e.g. because the PCH format changed between minor versions).

Anyway, I don't think there was a bug here. It seems like the tree was either checked out at a bad revision or something failed during the deps download. So, updating the checkout and doing a clean rebuild should solve it.

BTW, you don't need another script to do a clean rebuild. You can just rebuild everything by running [this command](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Windows.md#:~:text=build_win.bat%20%2Ds%3Dall) (plus any options for your config):
```
build_win.bat -s=all
```
That stated, as I mentioned in the other comment, I'll see about adding my deps tracking logic into `build_win.bat`. Because automating it should reduce this sort of confusion.

closed 2026-09-29 by prusaslicer-bot[bot]