prusa3d/PrusaSlicer #8113: Post Tool change gcode to support printers like Tool Changer (E3D) Or always set Z Axis after a tool change

traderhut, 2022-03-25:
Currently, if you have multiple tools in Tool Changer printer, it picks up one, and then drops the Z+5 so it can move the head safely, assuming that the generated GCode would simply move back to where it was about to print (X,Y AND Z).   Other slicers will move the X and Y, and then move the Z and start printing.

In the case of the Tool changer, you need to specifically move the Z Axis because switching tools does not leave the Z Axis set to the same as it was.

This also includes possible different Z offsets for the tools, for example, mine is 1.6mm lower for my 2nd tool than my first.  and it bumps up the Z Axis by 5mm to make sure it handles the change OK.   It moves back to correct Z Axis only on the next layer, so in the mean time it is dropping plastic on top of the part by about 3.6 mm above it.

I could fix it by executing code AFTER the tool change, but that isn't really the correct.

This can be easily fixed by simply adding a G1 ZXXX to the generated gcode right before it starts printing.

I was talking to someone else, who said this is the reason they had to dump PrusaSlicer for another one, in spite of the fact that they liked this package better.

I too like this better, love how it interfaces to the machine, so would prefer to be able to use this with multiple color prints.

fe60, 2022-03-25:
Post tool change custom gcode has been requested in #7892.

traderhut, 2022-04-06:
The Custom gcode won't work, here is why:

The issue that I specifically have with the E3D (Any RepRap with multiple Tools)  is that it needs to set the FULL LOCATION X,Y and Z Axis after a tool change - can not be solved with custom G-Code After tool change.  It sounds like it could, but it can't.  

Here is why:

You are drawing  at    X=30,Y=50, Z= 0.2   with T0
You are about to move to X=40,Y=70  (Z=0.2, but currently the move won't do that) , and draw using T1

You call the code for T1, which does <whatever> custom code, but eventually would issue a T1 to the printer.  The printer changes tools, but will also leave the current position at  X,Y = location of Tool changer, say X=-40, Y=250  (on a bed that is 200 wide), and Z=5.2   (it moves Z += 5 to avoid hitting anything)   (I Originally thought that this was the only issue, and could be fixed by simply removing the Z relative move... nope.)

Now, you are 'done' with the tool change, so the printer issues a G1 X40 Y70   and it starts printing in the air at Z=5.2.

Simply doing a Move to Z0.2  won't solve the issue, as then the printer will zoom across the print at print level to get from -40, 250 (or whatever random location) to where it is about to print, and if your nozzles are off by even 0.05mm in z height this causes it to smack the print at high speed as it zips across the top of it.   And the WILL be off by a lot more..

Due to the way the RepRap firmware code updates a tool offset doesn't move the physical location when it is changed, but rather changes the software location.

 I.E  say you have a tool that is 2mm lower than another one, so when you do the above you end up at Z=3.2, if you simply did a relative move to Z -= 5 before moving, you crash HARD into the print when you move back. 

  This is a weird aspect of the firmware (At first, I considered it might be a bug - but it isn't), but Let's say that your T0 offset is 0,0,0  (so, if you are at 1,1,1  then you move to 2,2,2  you move 1 in each direction),

 Now you issue a command when you switch tools to have the offsets be 0,0,2  - which indicates that you need to increase the Z by 2 in order to be in the same location.  Instantly, your CURRENT location becomes (from 2,2,2)  2,2, 0   - It doesn't move the physical location, just updates what that location is in software.  If you then do an absolute move to 2,2,2  (back to where you were)  it will do a Z +2 in order to be at the location.

This means that in order for the Tool Offsets to be effective you NEED an absolute Z (and X and Y for that matter) move following a tool change.

How to fix the issue:

Say, your next line starts at X40, Y70:

You issue:
T1
G1  X40 Y70    ; Move to where you are about to print  (Fixes any changes to X or Y software offsets to put the nozzle back)
G1 Z0.2           ; Put the Z axis back too

Instead of:
T1
G1 X40 Y70

The X and Y will apply the changes to the offsets of X and Y in addition to moving back (would happen on next move anyway)

Note the Z MUST be AFTER then XY change!  You want it to zoom across the print at Z +5...  Not moving down first.  And if you moved z first you would be skimming across the part, which kind of works but not good. You could be skimming over places that have already been printed (in another color) for this level, so the difference between the nozzle and your print is 0, or whatever your error is between your z measurements between T1 and T0, which means it will hit, or at the very least wipe one color over another)

(This is why Custom Post Tool change code won't fix it, that and I would need to know where it is about to move to, not where it is)

This is a pretty simple modification (I know, I made the change in the code locally and got it working in about 10 minutes - although my fix does one extra move per layer, which doesn't matter)

The move only needs to be done after a tool change.   (My solution was to set a flag when a tool change is done, and then on any XY move,  issue the move, then issue a move for the Z and reset the flag.)    My solution is a simple hack, I expect that there are cleaner solutions but this beats doing it with a post processing scan like I was doing before.

closed 2026-09-29 by prusaslicer-bot[bot]