prusa3d/PrusaSlicer #8114: Big gap between support and nearly horizontal perimeter

fedemox, 2022-03-25:
### Description of the bug

I'm printing a round static winged duct for forced ventilation. Because every wing is merged on the tube with fillets on every edge, supports are placed horizontally where fillets begin but don't follow the curvature, leaving a big air gap between the contact surface and the wing perimeter.

### Project file & How to reproduce

- lay the cylindrical object on one of the flat surfaces (defect is more evident on one side)
- enforce supports over 35 degrees
- slice with standard "0,10 mm DETAIL" profile

[duct.zip](https://github.com/prusa3d/PrusaSlicer/files/8350736/duct.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Microsoft Windows 8.1

### Printer model

Prusa Mini+

rtyr, 2022-03-28:
It seems to be caused by the fact that the bottom part of the fins is too thin to be printed with your extrusion width.

fedemox, 2022-04-07:
You're right. Thanks!
Unfortunately, using "Detect thin walls" and "Extra perimeters if needed" lead to a different issue
[image]
.

stepikovo, 2022-04-11:
@fedemox sometimes the only solution on how to print too thin walls is to use a smaller nozzle

closed 2022-04-11 by stepikovo