prusa3d/PrusaSlicer #8116: Freeze while slicing large model at 0.05mm layer height preset

Kyle-Falconer, 2022-03-26:
### Description of the bug

I'm trying to slice the [moon lamp](https://www.thingiverse.com/thing:3532973/files) at  100% infill, 0.05 mm layer height for printing. I'm able to slice it at 0.10mm layer height and observed that PrusaSlicer took about 6GB of video ram (shared memory on my system), after a long pause. When trying to slice at 0.05mm, I got an out of memory error and the program crashed.

The computer specs at the time were:
 Intel Core i5-7600K
16GB RAM
Nvidia Quadro K600 video card (1GB dedicated VRAM, 8GB shared memory with the system RAM)
Windows 10 64bit Professional

After upgrading the system memory to 48 GB RAM, this allowed for 24 GB of shared video memory and now the slicing never finishes when attempting the same model at 0.05 mm layer height. I let it run for nearly 2 hours and the slicing never finishes.

[image]

### Project file & How to reproduce

1. Open project file
2. confirm infill is 100%, with cubic
3. confirm layer height is set to 0.05mm ULTRADETAIL with PETG or PLA preset
4. click the "Slice now" button

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Windows 10 64 Professional

### Printer model

Prusa i3 mk3S

foreachthing, 2022-03-26:
Try to use the console: `prusa-slicer-console.exe --export-gcode moon_lamp_8_inches_refinement_level_360Y165.stl --layer-height 0.05 --load "c:\Users\USER\AppData\Roaming\PrusaSlicer-alpha\print\0.20mm NORMAL @ULTIMAKER2 - PPS.ini" --load "c:\Users\USER\AppData\Roaming\PrusaSlicer-alpha\printer\Ultimaker 2 - mod.ini" --load "c:\Users\USER\AppData\Roaming\PrusaSlicer-alpha\filament\PLA_Black_3DW.ini"` it peaked at around 5.4 GB RAM on my Windows 11 computer.
Forgot to stop the time ... I'll do it again and post an update.

Kyle-Falconer, 2022-03-26:
The CLI worked for me also, with similar results regarding RAM usage and execution time.

The command I used was:
```
prusa-slicer-console.exe --export-gcode moon_lamp_8_inches_refinement_level_360Y165.stl --layer-height 0.05 --load "C:\Users\kfalc\AppData\Roaming\PrusaSlicer\print\0.05mm ULTRADETAIL 100 infill.ini" --load "C:\Users\kfalc\AppData\Roaming\PrusaSlicer\printer\Original Prusa i3 MK3S & MK3S+ - copy.ini" --load "C:\Users\kfalc\AppData\Roaming\PrusaSlicer\filament\Prusament PLA - Grey.ini"
```

So, I guess this points to a bug in the GUI where it's needlessly using memory.

foreachthing, 2022-03-27:
I guess the GUI also wants to create the preview and has a hard time with it. I didn't wait for it to finish loading...
My gcode file was >1 GB. Maybe try another gcode viewer to see if it can handle that file.

foreachthing, 2022-03-27:
So, after some time (count not time it) it finished slicing. It has peaked at 15.4 GB of RAM.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]