prusa3d/PrusaSlicer #8117: Set extruder sequence for more than one object does not work

Bendermk6, 2022-03-26:
### Description of the bug

The "set extruder sequence" feature only allows you to set it when you want to print all objects on the build plate in the same set sequence.

If you want to print one object with set sequence and another in just one color, PrusaSlicer just ignore the set sequence at all.

Conclusion the "set extruder sequence" feature only works when all objects on the build plate are set to the same printer, if you have two objects with two different extruders set, the "set extruder sequence" isn't avaible.

It would be great to "set extruder sequence" object based and not build plate based.

### Project file & How to reproduce

Add two objects on the build plate.
Use "set extruder sequence" in preview mode and slice it.
Change extruder of one object and slice again -> Height Slider shows errors with the set sequence.
[image]
[image]

[extruder_sequence.zip](https://github.com/prusa3d/PrusaSlicer/files/8354815/extruder_sequence.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.1+win64-202203101054

### Operating system

Windows 10 (build 19044), 64-bit edition

### Printer model

Prusa MK3s+ MMU2s

rtyr, 2022-03-28:
Intended behavior, see https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4 for the info about the basic rules applied here.

Bendermk6, 2022-03-28:
Okey, maybe I just wanted to much 😄 

I found a workaround for my problem. I just created a STL with 300x300mm slabs with 0.2mm height and 100 replicates in with a distance of 0.2mm. Now I can use the STL as a modifier for my second object on the build plate.

closed 2022-03-28 by Bendermk6