prusa3d/PrusaSlicer #8118: Extrude a bit more before the end of the print job

laptop-laptop, 2022-03-26:
**Is your feature request related to a problem? Please describe.**

In visualization of PrusaSlicer, the nozzle moved across the entire final top layer before ending the print job. Everything looked fine to me. However, in actual prints of different designs, the last bit (about 3-4 mm in length) showed up in different color (dull looking) than the rest of the shinning looking prints. Is this normal? Not sure if it is a problem with i3MK3S+ or the slicer (2.4.1 and before).

**Describe the solution you'd like**
Perhaps add a feature to extrude a bit more filament at the last 3-4mm before ending the print job?

rtyr, 2022-03-26:
> However, in actual prints of different designs, the last bit (about 3-4 mm in length) showed up in different color (dull looking) than the rest of the shinning looking prints. Is this normal? Not sure if it is a problem with i3MK3S+ or the slicer (2.4.1 and before).

Please contact our tech support via live chat at www.prusa3d.com.

closed 2022-03-26 by rtyr

laptop-laptop, 2022-03-26:
Are you an employee of Prusa? Tech support via live chat cannot answer my questions!