prusa3d/PrusaSlicer #8121: Parts split in to multiple parts when using modifiers to change perimeters.

zedfactorial, 2022-03-27:
### Description of the bug

I have talked about this issue with LiveChat and they told me to post it here

I'll give the simplest possible example. Let's say that we have a box with 2 perimeters. We want to reinforce a corner with an extra perimeter so we use a modifier cylinder.

[image]

What ends up happening is that the slicer pretty much treats the part region out of the modifier and the region in the modifier as 2 separate parts. This is the result:
[image]
[image]
[image]
It produces a nasty line on the sides and top of the model. It also introduces a weak point as the interface between two regions can be quite poor.

This is what I would expect the "correct" behaviour to be. Blue are the original perimeters and green is the extra one added by the modifier.
[image]
Have it create just one extra perimeter within the modified region. This part would not have the nasty line on the side and would avoid the problem of the poor mechanical interface within the part.

It's an even bigger issue with thin walls.
When slicling this file:
[image]

The tentacle-like protrusion get sliced with tiny specs of internal perimeter that I need to remove
[image]

Using a simple box modifier (in reality I would use a custom toroid for this) 
[image]

The modifier has 1 perimeter, 0% infill and no top and bottom layers. This removes the junk. but this introduces issues
[image]
The RED arrows show the region out of the box. As expected still has the junk. 
The GREEN arrows show that the area in the box is fixed as expected. 
The BLUE arrows show the unintended side effect of splitting the part and treating it as separate parts which pretty much ruins the part.

I know there are workarounds. I usually split the parts in CAD so I have more control and don't have to fiddle around with modifiers. However, this is something I am encountering often and having to work around it is annoying and takes time so it would be very much appreciated if it could be fixed.

### Project file & How to reproduce

1) Add any part into the slicer
2) Create a modifier region that partially covers the part
3) Set the modifier to have a different number of perimeters than the rest of the part.
4) Slice

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1 and all previous versions that I have tried from the last 2 years.

### Operating system

Window 10

### Printer model

MK3S (also MK3S with MMU2) but this is a slicer issue

rtyr, 2022-03-28:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3370, https://github.com/prusa3d/PrusaSlicer/issues/3317, https://github.com/prusa3d/PrusaSlicer/issues/3846, etc.

The way the modifier volumes work with current implementation is they split the object into pieces, which are sliced independently. It may be improved in the future. 

I am closing this as a duplicate.

closed 2022-03-28 by rtyr

zedfactorial, 2022-03-28:
I have been told more than 3 separate times to post this here by the live chat staff. 

If this is such a prevalent issue why has there not been a fix for multiple years and why is the live chat staff still not aware of this?

stepikovo, 2022-03-30:
> why is the live chat staff still not aware of this?

Thank you for the info, we will let everybody know.

marcelismus, 2022-08-01:
+1 for "fixing"