prusa3d/PrusaSlicer #8122: When you change setting with a blank project and go to exit it will ask you to save blank project, if you don't "save", prusa slicer wont' shut down

recurvehunt, 2022-03-27:
### Description of the bug

When you change setting with a blank project and go to exit it will ask you to save blank project, if you don't "save", prusa slicer wont' shut down.

### Project file & How to reproduce

Launch slicer. Change a setting. Save the setting change. Try and close the slicer, without saving a blank project 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1 + win 64

### Operating system

Windows 10

### Printer model

custom

rtyr, 2022-03-28:
Technically, the project was changed, but maybe there is some room for improvement.

buzzysale, 2022-04-05:
This is also documented in #8076

rtyr, 2022-04-05:
That is correct, I misunderstood the description of the issue. I am not able to reproduce the issue though.

MrDixioner, 2022-04-05:
I will describe in more detail. You may be able to reproduce the issue.
1. I opened the program and decided to make changes, for example, change the temperature of the table.
2. After making changes, an asterisk appears to the left of the project name (indicated by an arrow).
[image]
3. Then, I want to exit the program. The following dialog appears:
[image]
I have only one way out: click Yes to save the project (which I don't need), because if I click No, the program won't close.

buzzysale, 2022-04-05:
This occurs for *any* project that has an empty build plate.  

Here's another more problematic case where the bug causes real trouble:

Work very hard on a complex project. Save it, close PS, re-open the project, make a tweak you do not like, delete the model from the bed, then try to close the project.  It will not close and the only option is to overwrite your un-tweaked project with a blank bed or save it as a different project.

MrDixioner, 2022-04-05:
Yes, that's right, I have the same problem.

buzzysale, 2022-04-05:
As stated in #8076 the behaviour of the No button should be "close without saving" and I suspect the No button is set to just cancel the save dialog.

MrDixioner, 2022-04-05:
Most likely it is. Let's wait for the patch.

lukasmatena, 2022-04-13:
Should be fixed in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). Let us know if it isn't. Closing the issue.