prusa3d/PrusaSlicer #8123: PrusaSlicer process is gobbling up memory on Macbook pro (Apple M1 Pro chip)

dccam, 2022-03-27:
### Description of the bug

Slicer seems to be using more and more memory.  Eventually my computer throws up an error that memory is getting critically low and makes me close programs.

I don't know if previous releases used up memory the same but I never saw this warning before
[image]

### Project file & How to reproduce

This kind of memory usage seems to be happening with all projects.  PrusaSlicer's memory usage shoots up when in use but even when I don't have it in the foreground the usage continues to creep up

System and version details are below
PrusaSlicer
Version:   2.4.1+arm64
Build:     PrusaSlicer-2.4.1+arm64-202203101039

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 12.0.1 (Build 21A559)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   2.1 Metal - 76.1
Vendor:       Apple
Renderer:     Apple M1 Pro
GLSL version: 1.20

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1+arm64

### Operating system

  macOS Version 12.0.1 (Build 21A559)

### Printer model

PRusa Mk3s

bubnikv, 2022-03-28:
This is quite unusual. Our tester tests primarily on a M1 based MacBook Air.
Do you slice some large projects?

bubnikv, 2022-03-28:
@FidelCapo tried to reproduce on his M1 with no success. Would you please share your project files?

n8bot, 2022-03-28:
Is it intended to use the Arm64 build, instead of the MacOS universal build?

dccam, 2022-03-28:
@FidelCapo Here is a current file that I just tested and saw the same issue...though it's happening with all my PrusaSlicer files.
[GH_bottle opener.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8364735/GH_bottle.opener.3mf.zip)

dccam, 2022-03-28:
> Is it intended to use the Arm64 build, instead of the MacOS universal build?

I also wasn't sure about the build so I redownloaded the universal build and installed it over my current version.  The information about is all from PrusaSlicer after reinstalling the universal build

dccam, 2022-03-28:
> This is quite unusual. Our tester tests primarily on a M1 based MacBook Air. Do you slice some large projects?

Sometimes I slice large files but this is happening with all files.  The project I attached above is only 55kb

dccam, 2022-03-29:
I found this article and it sounds like it's a known issue with macOS Monterey.

The first thing they suggest is disabling custom cursor size and color.  That is something that I changed recently.  I'll change the cursor setting and report back.

dccam, 2022-03-29:
Confirmed! My apps are much better behaved since I changed back to the default cursor size and color

bubnikv, 2022-03-29:
Great, thanks.