prusa3d/PrusaSlicer #8125: Add option to disable automatic color change

CheeseTater, 2022-03-28:
**Is your feature request related to a problem? Please describe.**
At times the automatic color change will apply color changes to models where the feature is unwanted. This has happened to me with several models recently and my only recourse to either remove the color changes manually or use another slicer.

**Describe the solution you'd like**
An option to disable automatic color change

**Describe how it would work**
A toggle in preferences that disables automatic color change

**Describe alternatives you've considered**
An alternative would be an option in layer height area of preview view that removes all color change settings similar to "Set auto color changes"

**Additional context**
Add any other context or paint drawing about the feature request here.

rtyr, 2022-03-28:
You can remove all color changes by using this icon:

[image]

Closing.

closed 2022-03-28 by rtyr

CheeseTater, 2022-03-28:
"Discard all custom changes"  removes all changes not just those regarding color changes

rtyr, 2022-03-28:
Color changes, pauses and custom gcodes. I think it is sufficient way how to remove accidentally added automatic color changes. It should be quite rare since user have to confirm adding automatic color changes.

CheeseTater, 2022-03-28:
The automatic color changes were being applied to some models for me without confirmation as soon as I went to slice them, but that would seem to be a bug then. Next time it happens I'll open a new issue so I can provide proper details.

rtyr, 2022-03-28:
It should never happen, PrusaSlicer should only show notification/suggestion which user have to confirm.