prusa3d/PrusaSlicer #8126: segfault, assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

fiechr, 2022-03-28:
### Description of the bug

Since a few days I get this error, when trying to start the built application from the master branch.

Older versions (maybe about 2-3 weeks old) work fine.

OS is Manjaro Linux.

I know these builds aren't officially supported, but maybe it's an easy fix and/or also relevant for the official builds.

```
[2022-03-28 13:14:27.550805] [0x00007fbf21444840] [trace]   Initializing StaticPrintConfigs

(process:13314): Gtk-CRITICAL **: 13:14:27.557: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:13314): Gtk-CRITICAL **: 13:14:27.557: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(process:13314): Gtk-CRITICAL **: 13:14:27.557: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
[1]    13314 segmentation fault (core dumped)  prusa-slicer
```

### Project file & How to reproduce

n/a

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

g888f45c0d

### Operating system

Manjaro Linux

### Printer model

n/a

Salamandar, 2022-03-29:
Maybe add "[master]" in your issue title, so that developpers know it's not on a released version of their software :)

lukasmatena, 2022-03-30:
@fiechr Your report lacks any information about how PS was built. All I can say is that my master build which I built today using the static dependencies built using the deps script (as described in https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md) is running fine. Did you build it the same? Can you still reproduce it with current master?

fiechr, 2022-03-30:
Current master still produces the same error.

Since I'm using the package from the Arch AUR, this is how it is build currently:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=prusa-slicer-git

Salamandar, 2022-03-30:
If there's an issue in the packaging, I'm the one to blame :)

lukasmatena, 2022-03-30:
@fiechr I reproduced the problem with static build against wxWidgets 3.0 (just by setting `SLIC3R_WX_STABLE` to true). It crashes in wxWidgets after a recent change, I don't know yet if the problem is in wxWidgets or in the way that we call them. I will investigate.

@Salamandar We only build statically against a patched version of wxWidgets 3.1.4, so linking against stable and ancient wx3.0 is generally discouraged and produces something else that we test, but it probably does not count as packaging issue...

lukasmatena, 2022-03-30:
I attempted to fix it with 8004e19. Would you please confirm? The crash was inside wxWidgets `wxExecute` call when called with `wxEXEC_SYNC` but without `wxEXEC_NOEVENT`. Not pumping the event queue when waiting for the child to finish fixed the problem on my setup.

It may quite well be a bug in wxWidgets that is already fixed in 3.1. Quite likely this one: https://github.com/wxWidgets/wxWidgets/issues/16661. It mentions a race condition manifesting when the child exits too fast, which is to be expected in this case. I don't know if setting `wxEXEC_NOEVENT` fixes it, or just coincidentally makes another case of the race condition win. Hopefully the former.

This is the ASAN backtrace I got, in case it is needed in future:
```
==12602==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000018 (pc 0x7fd1b50ed104 bp 0x7fd1b55b5b20 sp 0x7ffed028dbf8 T0)
==12602==The signal is caused by a READ memory access.
==12602==Hint: address points to the zero page.
    #0 0x7fd1b50ed103  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x2e1103)
    #1 0x7fd1b4f93cab  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x187cab)
    #2 0x7fd1b4fb4156  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x1a8156)
    #3 0x7fd1b4f9f341  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x193341)
    #4 0x7fd1b4fb4081  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x1a8081)
    #5 0x7fd1b4fb40dc  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x1a80dc)
    #6 0x7fd1b4f9fd75  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x193d75)
    #7 0x7fd1b48181a0 in g_type_create_instance (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x3b1a0)
    #8 0x7fd1b47f734c  (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1a34c)
    #9 0x7fd1b47f8b44 in g_object_new_with_properties (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1bb44)
    #10 0x7fd1b47f96f0 in g_object_new (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1c6f0)
    #11 0x7fd1b4fbca6e  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x1b0a6e)
    #12 0x7fd1b519bb1c  (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x38fb1c)
    #13 0x7fd1b48181a0 in g_type_create_instance (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x3b1a0)
    #14 0x7fd1b47f734c  (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1a34c)
    #15 0x7fd1b47f9377 in g_object_new_valist (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1c377)
    #16 0x7fd1b47f96cc in g_object_new (/lib/x86_64-linux-gnu/libgobject-2.0.so.0+0x1c6cc)
    #17 0x7fd1b5b707d2  (/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0+0x2777d2)
    #18 0x7fd1b5b81eeb in wxCursor::InitFromStock(wxStockCursor) (/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0+0x288eeb)
    #19 0x7fd1b5cbbb30 in wxStockGDI::GetCursor(wxStockGDI::Item) (/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0+0x3c2b30)
    #20 0x7fd1b5bb97b1 in wxGUIAppTraits::WaitForChild(wxExecuteData&) (/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0+0x2c07b1)
    #21 0x7fd1b61668d7 in wxExecute(char**, int, wxProcess*, wxExecuteEnv const*) (/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0+0x1e98d7)
    #22 0x7fd1b616847f in wxExecute(wxString const&, int, wxProcess*, wxExecuteEnv const*) (/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0+0x1eb47f)
    #23 0x7fd1b6190883  (/lib/x86_64-linux-gnu/libwx_baseu-3.0.so.0+0x213883)
    #24 0x555ca6101715 in Slic3r::GUI::desktop_execute_get_result(wxString, wxArrayString&) /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/GUI/GUI.cpp:561
    #25 0x555ca59c69a9 in get_downloads_path /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/Utils/AppUpdater.cpp:88
    #26 0x555ca59c6ea5 in Slic3r::AppUpdater::priv::priv() /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/Utils/AppUpdater.cpp:155
    #27 0x555ca59cff19 in Slic3r::AppUpdater::AppUpdater() /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/Utils/AppUpdater.cpp:512
    #28 0x555ca504a814 in std::_MakeUniq<Slic3r::AppUpdater>::__single_object std::make_unique<Slic3r::AppUpdater>() (/home/lukas/PrusaSlicer/PrusaSlicer/build/Debug/src/prusa-slicer+0x26d5814)
    #29 0x555ca4feb48b in Slic3r::GUI::GUI_App::GUI_App(Slic3r::GUI::GUI_App::EAppMode) /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/GUI/GUI_App.cpp:824
    #30 0x555ca4fdc6bb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) /home/lukas/PrusaSlicer/PrusaSlicer/src/slic3r/GUI/GUI_Init.cpp:42
    #31 0x555ca31e2490 in Slic3r::CLI::run(int, char**) /home/lukas/PrusaSlicer/PrusaSlicer/src/PrusaSlicer.cpp:616
    #32 0x555ca31e73ce in main /home/lukas/PrusaSlicer/PrusaSlicer/src/PrusaSlicer.cpp:872
    #33 0x7fd1b43570b2 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x240b2)
    #34 0x555ca31d979d in _start (/home/lukas/PrusaSlicer/PrusaSlicer/build/Debug/src/prusa-slicer+0x86479d)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/lib/x86_64-linux-gnu/libgtk-3.so.0+0x2e1103) 
==12602==ABORTING

closed 2022-03-30 by lukasmatena