prusa3d/PrusaSlicer #8127: No tool selection Tn sent at the start of a job

traderhut, 2022-03-28:
### Description of the bug

If you select a part to print, specify Extruder 1 (Tool 1), it doesn't actually send the T1 command to the printer unless you switch to another tool and switch back.  It assumes that you started the job with the correct tool selected.

It would make (more) sense to do that if you had selected 'default' for the tool, but if you specify tool 1, it should send a T1 before starting the job.   Some printers (E3D for example) do not have a 'default' tool that is active when you start the job, and if you add commands to home the axis, it will not have any tool selected.  

This leaves you with creating Pre-print custom gcode that selects the tool, and you have to be sure to change that to whatever tool it uses first.  (If you had T0 and T1, it would need to start printing with the right tool)

Should be an easy (trivial) fix.

### Project file & How to reproduce

Create a project, drop a model on it, select Extruder 1 as the extruder to use, You should (but don't) see a T1 at the startup code for the print.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1 WIndows

### Operating system

Windows

### Printer model

Tool Changer, E3D - but it could be any printer

rtyr, 2022-03-29:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/3313, https://github.com/prusa3d/PrusaSlicer/issues/1796.

You need to add `T[initial_tool]` to the start g-code.

Can we close the issue?

rtyr, 2022-04-05:
No response, same as https://github.com/prusa3d/PrusaSlicer/issues/3313, https://github.com/prusa3d/PrusaSlicer/issues/1796. Closing.

closed 2022-04-05 by rtyr