prusa3d/PrusaSlicer #8129: Please add the releases to winget, more rapidly/steadily

cloudrootab, 2022-03-29:
**Is your feature request related to a problem? Please describe.**
The versions published to the winget repo is still 2.4.0, even after several weeks since the release of 2.4.1.

**Describe the solution you'd like**
Have at least the stable releases, either automatically or quickly following after release, be published to the winget repo.

**Describe how it would work**
You've already published a couple of versions to winget. However, these are older versions, lagging behind with several weeks.

**Describe alternatives you've considered**
Manually downloading, for sure. But it would be convenient to not have to do that. 

**Additional context**
None

fe60, 2022-03-29:
According to the PrusaSlicer documentation, the official download options are the download section of the Prusa3d web page and the Github releases [1].

All other options are maintained by enthusiasts supporting PrusaSlicer and a certain distribution channel.
The documentation states about Linux, all distribution specific packages _are not officially tested by PrusaResearch thus not guaranteed to be working or up to date_ [2].

> You've already published a couple of versions to winget.

None of the commiters in the winget-pkgs repo seems to be related to Prusa Research [3].

[1] https://help.prusa3d.com/en/article/download-prusaslicer_2220
[2a] https://help.prusa3d.com/en/article/install-prusaslicer_1903/#linux
[2b] https://github.com/prusa3d/PrusaSlicer/wiki/PrusaSlicer-on-Linux---binary-distributions
[3] https://github.com/microsoft/winget-pkgs/commits/master/manifests/p/Prusa3D/PrusaSlicer

cloudrootab, 2022-03-29:
> None of the commiters in the winget-pkgs repo seems to be related to Prusa Research [3].

I tip my hat to your digging-skills. It was obviously more thorough than mine. However, I would then like to propose that Prusa take ownership over that "release-channel" and implement it into their workflow. But yes, this is a feature request. But from a certain point of view it does make sense that Prusa would be the ones holding the reigns on the winget-repo.

cloudrootab, 2022-11-03:
I would like to add a small discrepancy between the documentation, and acctual behaviour. Whereas the docs may say that the official download places are via the website. Acctually clicking the "Software Releases" from within the slicer, takes the user to the Github release pages. 

Obviously these options aren't mutually exclusive. And can happilly exist next to, and along side each other. My point here being that there obviously are more than ONE path to the binaries. So there could be a third as well?

Just my two cents.

closed 2026-09-29 by prusaslicer-bot[bot]