prusa3d/PrusaSlicer #8134: Multiple retracts on single infill lines not needed (so it seems)

kevinmichaeldudley, 2022-03-30:
### Description of the bug

The slicer is adding MANY retracts to a circular path of lines when only ONE is needed.
If the perimeter extrusion width value is at 0.4 mm then MANY multiple retracts occur on the infill path.
If the perimeter extrusion width value is at 0.401 then 8 retracts occur.
If the perimeter extrusion width  is at 0.41 then only ONE retract occurs.

It appears that the slicer is "jumping" between applying an "infill" line and a "perimeter" line.
This causes MANY extra retracts and adds 25% more time to a part similar to this.

Maybe:  The internal coding which decides to "switch" to an infill or a perimeter line perhaps needs an "over under overlapping threshold" to prevent this "jumping" instead of a direct greater/less than condition.

Just thought I should report it as it does add a significant amount of time.  I have seen this on other similar parts.

Just change the perimeter extrusion width to see the issue.
[Multiple Retract Report.zip](https://github.com/prusa3d/PrusaSlicer/files/8381714/Multiple.Retract.Report.zip)

Shows single retract applied if using perimeter extrusion width of 4.1
[image]

Shows multiple retract applied if using perimeter extrusion width of 4.0
[image]

### Project file & How to reproduce

Just change the perimeter extrusion width to see the issue.
[Multiple Retract Report.zip](https://github.com/prusa3d/PrusaSlicer/files/8381714/Multiple.Retract.Report.zip)

Shows single retract applied if using perimeter extrusion width of 4.1
[image]

Shows multiple retract applied if using perimeter extrusion width of 4.0
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 10

### Printer model

Mk3s

kevinmichaeldudley, 2022-09-04:
Tested and appears to be fixed.

kubispe1, 2022-09-04:
Thank you for approval

closed 2022-09-04 by kubispe1