prusa3d/PrusaSlicer #8136: Reduce Stringing due to travel moves after layer change

gary-wlg, 2022-03-30:
When printing with filament types that tend to ooze out of the nozzle (eg. TPU) you may set a filament override to reduce to the Lift Z to 0 for retractions so that filament doesn't ooze out of the nozzle during travel moves.   This when paired with the setting "Avoid Crossing Perimeters" can eliminate a lot of stringing.  

The issue is that on layer change, the sequence of events is to first increase the height of the nozzle to change layer, then perform a travel move to the start location for the layer, then start extruding.  This travel move is performed with the nozzle basically lifted from the previous layer, which can cause a lot of stringing. This stringing can be eliminated if the sequence was changed so that when it is time for a layer change, the first step is to perform the travel move to position the nozzle at the start position for the next layer, then increase the nozzle for the layer change, then start the extrusion moves.

I've attached an image that shows the events.

- The red arrow is the start of the new layer and where the layer change occurs increasing the nozzle height.
- Next the travel move is performed (the blue line)
- Finally the extrusion moves start.

In this case you end up with a lot of stringing where noted in the image.  By changing the sequence of events so that the layer change nozzle increases height after the travel move, the stringing is completely eliminated.

It would be good if either a new option could be available to "Perform Travel Move before Layer Change" or some other way to have this happen.

[image]

SeanLMcCullough, 2022-05-21:
I came across this same issue when dialling in eSun PLA+ stringing. The GCode generated by the teaching tech website had my retraction dialled perfectly, no stringing or blobs. Their GCode is based on Simplify3D. Analysing the GCode shows it printing the next layer as near as possible. This results in a test that goes right, left, left, right, right, left, etc. Analysing the GCode from PrusaSlicer seems to go left, right, left right, left right, etc. Same problem that the Z lift before travel effectively produces stringing as though it was a z-hop.

Maybe there's a good reason for the behaviour, but I see two ways of mitigating the issue. Either lifting Z after travel unless Z hop is enabled, or begin the next layer at the nearest neighbour rather than travelling back to a part origin.

npburns224, 2022-09-27:
I’ve spent the last few days dialing in a few different spools of PLA only to discover this issue myself. Stringing is nonexistent between normal travel moves, but still presents itself during a layer change due to the z lift. No combination of slicer settings seems to alleviate the issue.

closed 2026-09-29 by prusaslicer-bot[bot]