prusa3d/PrusaSlicer #8137: Infill "line 45"

livius2, 2022-03-31:
**Is your feature request related to a problem? Please describe.**
Currently we have infill "line", but it is angled to the plate by 45 degrees.
And if we need that long lines go along the entire object, we must rotate it on the plate by 45 degrees.
But then the object can be angeled along both axis and the printer must move on x and y axis.
I like to avoid move along MK3S+ y axis in this infill at all.

[image]

**Describe the solution you'd like**
Please add modified current line infill to be angled by 45 degrees. Maybe with name "Line 45" (just proposition).

neophyl, 2022-03-31:
Why aren't you changing the infill angle settings to adjust the infill ?  Print Settings>Infill>Advanced>Fill Angle.  Its a yellow setting so you need to be in advanced or expert mode to see it.

livius2, 2022-03-31:
Because I did not notice that such an option exists :) 
Ty for pointing me for this so important option :)