prusa3d/PrusaSlicer #8138: Explicit control over slicing tolerance

ben-kaye, 2022-03-31:
Like Cura, it would be great if we could get control over the logic for the position of the edge when dealing with a sloped model. It is very concisely explained here:
https://support.ultimaker.com/hc/en-us/articles/360012614559-Experimental-settings#slicing%20tolerance

Three options: inclusive, exclusive and median. 
*Exclusive is the maximal set that would fit within the model. 
*Inclusive is the minimal set that would fit around the model.

Alternatively if the slicing logic was adequately explained so that tolerances could be designed around this.

poken1151, 2023-05-15:
Also looking to get this feature in as well, would be lovely to have as an additional tool for nozzle and print mismatch.

tak-21, 2023-09-27:
Just learned about this topic while printing a bolt. It would be really nice if this feature would become available as it would prevent me from rescaling parte of thhe to be printed model to compensate the described effect.

tripletto, 2023-10-01:
This would be awesome and is kind of essential anyways.

AJolly, 2024-03-04:
Its the one thing I love cura for.

eskimgv, 2024-07-30:
Is there any update about this topic?

eskimgv, 2024-08-20:
> Is there any update about this topic?

That was the excactly my problem and finaly i solved last night :) https://github.com/prusa3d/PrusaSlicer/issues/3062#issuecomment-2297698540

tripletto, 2026-09-20:
Still want the feature though.

It's extremely useful to do perfect friction fits without needing to actually design any tolerances into the model.

You can just model a 10 mm hole and a 10 mm dowel set slicing to exclusive (Cura's term) and it fits.

Orum, 2026-09-20:
Yeah but it needs to be a "new user" that comments.  In all likelihood this will be closed and we'll just have to open a new issue for it.  Gotta love the thoughtless bot programming...

drewoprea, 2026-09-20:
I've been watching this and hoping it gets added to PrusaSlicer since late 2023.

UxuginPython, 2026-09-20:
Here's a comment. Don't close the issue.

Jan-Soustruznik, 2026-10-01:
Hello, @tripletto , thank you for update, According to new [policy](https://github.com/prusa3d/PrusaSlicer?tab=contributing-ov-file#issue-tracker-policy) I closing this feature request and converting it into a Discussion thread to hear what the community thinks about. Thank you, and have a nice day.