prusa3d/PrusaSlicer #8141: 0 gap mesh/faces treated as combined object

sarf2k4, 2022-04-01:
### Description of the bug

I have a 0 gap mesh/face design, and upon slicing I would like to use **negative xy compensation** in order to make the whole thing print in place. The slicer just gone ahead and combined any 0 gap meshes/faces as intersects.

I am expecting that slicer is not going to _combine_ them together and treat them as different object that have **xy compensation** applied to all xy sides of the mesh.

Image below exhibits 2 different model, one on the right; 0 mesh/face gaps, while on the left has 1 micron or less gaps, that slicer behaved just the way I expected it to be
[image]

I wanted to have a 0 gap mesh/faces on my object if that's even possible then utilize **xy compensation**

### Project file & How to reproduce

Use **negative xy compensation** 

[New Test Print.zip](https://github.com/prusa3d/PrusaSlicer/files/8394273/New.Test.Print.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.1+win64-202203101054

### Operating system

Operating System: Windows System Architecture: 64 bit Windows Version: Windows 10 (build 19043), 64-bit edition

### Printer model

prusa mk3s mmu2

rtyr, 2022-04-01:
Select the model and use "Split to objects" option. It should solve your issue.

[image]

sarf2k4, 2022-04-01:
> Select the model and use "Split to objects" option. It should solve your issue.
> 

Thank you for the reply, it did solve the problem, I forgot to enable **support on buildplate only** on the initial 3mf file and the following happened

[image]

rtyr, 2022-04-01:
Yes, that is expected with separate objects. I suggest painting supports only where you need them or modify the object so there is no 0 gap.

rtyr, 2022-04-01:
An example:
[New Test Print2.zip](https://github.com/prusa3d/PrusaSlicer/files/8396792/New.Test.Print2.zip)