prusa3d/PrusaSlicer #8142: no support auto generated

ristein, 2022-04-01:
### Description of the bug

No support will be generated, even I have an 80mm bridge.
[image]
[image]

### Project file & How to reproduce

[no_support.zip](https://github.com/prusa3d/PrusaSlicer/files/8396556/no_support.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.1+win64

### Operating system

Windows 10 21H2

### Printer model

Creality Ender 5 plus

rtyr, 2022-04-01:
You have the "Dont't support bridges option" enabled, disabling it will generate support there.

[image].

Closing.

closed 2022-04-01 by rtyr

ristein, 2022-04-01:
thanks a lot!
strange that this feature is on by default (at least with my printer).
would be nice to have a setting for "minimum bridging distance that needs support"

rtyr, 2022-04-01:
> would be nice to have a setting for "minimum bridging distance that needs support"

There is already request for that https://github.com/prusa3d/PrusaSlicer/issues/3859