prusa3d/PrusaSlicer #8144: Fill bed with instances bug

5jvm0u4, 2022-04-01:
### Description of the bug

I loaded a multiple object file, split it into objects, and want to use fill bed with instances function but it doesn't work.
But when I use add instances, it worked.

### Project file & How to reproduce

[Fill Bed With Instances Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/8397329/Fill.Bed.With.Instances.Bug.zip)

(Fig. 1)When one opens the file I attached, try to use fill bed with instances with the big part, to me, that doesn't work.
(Fig. 2)But it is physically possible to fill the bed with 4 instances, and when applying fill bed with instances to the small part, it works. (Fig.3)

[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1 final release

### Operating system

Windows 10 21H2

### Printer model

Original Prusa i3 MK3s

fe60, 2022-04-01:
Moving the small object into a corner of the virtual print bed and "Fill bed with instances" arranges two of the large objects on the bed. Place the small object outside of the print area and PrusaSlicer fills the bed with four of the large objects.

stepikovo, 2022-04-05:
_Fill bed with instances_ uses Arrange tool but only for the selected object/s, in this case only the big one, so it won't move the small one aside, however, there really is a bug, it should add at least one more of the big objects, see the screenshot.  
As @fe60 mentioned, you can move the small one manually and then it can fill the bed.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]