prusa3d/PrusaSlicer #8146: V2.4.1 Crashes when opening 3mf file for printer that is not configured

pickworthi, 2022-04-01:
### Description of the bug

What happens:
Prusa Slicer is configured on my LInux PC for a Prusa Mini+ (with four nozzle sizes).
From and open instance of Prusa Slicer, I click "Open Project" or "Import - ..3mf.." and select PreparedTempTowers.3mf.
The file is attached in Project file section, it is available on [https://www.printables.com/model/154630-temp-tower-temperature-test-stringing-and-bridging/files](url)
Prusa Slicer crashes after loading the file.

Workarounds:
If I double click on the file, or pass it as a parameter to Prusa Slicer via the command line, or drag and drop the file onto a running instance, the 3mf file is loaded, and Prusa Slicer informs me that temporary profiles have been imported for I3 MK3S and Prusament PETG.

Expected result:
Using "Open" or "Import" should not crash, but work as for command line or drag and drop invocation and import the missing profiles.

I have attached my config export:
[config.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/8399999/config.ini.zip)
.
Sorry, I can't get a screenshot of the program crashing, it just disappears.

### Project file & How to reproduce

[PreparedTempTowers.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8399944/PreparedTempTowers.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.1+linux-x64-GTK2-202203101048

### Operating system

 Linux 5.17.1-gentoo-r1 x86_64

### Printer model

Prusa Mini+

lukasmatena, 2022-04-05:
Thanks. I managed to reproduce the crash. I will have a look at it.

lukasmatena, 2022-04-05:
This is a duplicate of #8078 and #8097. The fix will be released soon in 2.4.2 patch release. Closing.