prusa3d/PrusaSlicer #8147: M600 in Artillery Genius continues printing but does not release filament

Fory360, 2022-04-01:
### Description of the bug

m600 in artillery genius

Hello, I generated the gcode in prusaslicer and sent it to octoprint, it asks for the filament change, the extruder moves away, I change the filament and next.... it goes to its position where it stopped, and the printing continues but it does not release the filament through the nozzle , at the end I check if there is a nozzle obstruction but there is not, the filament comes out perfectly.
Simplify3D works fine and Ultimaker Cura but PrusaSlicer doesn't.

### Project file & How to reproduce

[Artillery_Genius_71_Ruedas_Extension_0.16mm_PLA_Genius_8h26m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/8400725/Artillery_Genius_71_Ruedas_Extension_0.16mm_PLA_Genius_8h26m.gcode.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.3 x64

### Operating system

macOS 11.6.2

### Printer model

Artillery Genius

rtyr, 2022-04-01:
PrusaSlicer only adds M600 command to the gcode, the rest is handled by the firmware. I checked your gcode, M600 is there as expected. My guess is that your issues are related to the firmware or Octoprint.

I don´t see any issue related to PrusaSlicer. As you can see on the screenshot, the g-code is ok after M600.

[image]

Fory360, 2022-04-02:
check the gcode too, it's strange but notice the small stl prin

that's prusaslice.

[image]

that's Simplify3D

[image]

I find Prusaslicer more comfortable for me, but I don't believe the color changes correctly

rtyr, 2022-04-02:
Simplify doesn't have such feature as far as I know.

It seems you have not enough pressure in the nozzle after returning from M600. You can modify the color change gcode so there is some additional priming after M600.

Fory360, 2022-04-02:
if you have it but by script {REPLACE "; layer 50," "M600\n; layer 50,"}, could you explain what I have to add to make it work correctly

rtyr, 2022-04-02:
You can add additional line to your color change gcode in printer settings (custom gcode section).

For example:
M600
G1 E1.9 F1500

It will prime the nozzle with additional 1.9mm of filament after returning from M600. This is just example value.

That Simplify example is just gcode post-processing, which ignores retraction state, so it probably does deretraction, even though previous retraction was performed with different filament. So it basically does by accident what I described in that example.

Fory360, 2022-04-02:
thanks i will try your solution

closed 2022-04-05 by stepikovo

fontenele, 2022-04-18:
The method `GCode::process_layer()` have the own lifecycle, and all layers starts with retraction.
The M600 (change color) do not needs the retraction when start printing.
Tried the option **Extra length on restart** but this not affected the final gcode. Tried override this at **Filament Settings** and failed too.
Blob bug was fixed but created this issue. [#6362](https://github.com/prusa3d/PrusaSlicer/issues/6362)

All layers (not the first) starts with retraction:
[image]

#6362 fix
[image]

**Ugly solution:**

- **Remove the first line after the M600 at gcode. (G1 E-....)**

rtyr, 2022-04-19:
The retraction is correct (you can see in the gcode, that there is also the deretraction before printing). 

You can add additional priming (which will be different for each printer/fw):
https://github.com/prusa3d/PrusaSlicer/issues/8147#issuecomment-1086699008

fontenele, 2022-04-19:
```
G1 Z3.08 F9000
;AFTER_LAYER_CHANGE
;3.08
;COLOR_CHANGE,T0,#2EBB2A
M600
G1 E1.9 F1500 ; Custom GCODE exampled
G1 E-5 F3600 ; but retract before start remains here
G1 X106.585 Y71.268 F9000 ; start printing without release filament because retracted -5 at 3600 at line above
G1 X106.532 Y71.472
G1 X106.461 Y71.97
G1 X106.44 Y72.487
...
```

rtyr, 2022-04-19:
The part you marked as "start printing" is not printing, it is only travel. Retraction before travel is expected and intended.

fontenele, 2022-04-19:
You are right, its just travel.
But why need to retract if we purged filament before start new color? the nozzle is ready to print and the filament at a good point.
If was just a sample layer, ok to make a retract.

rtyr, 2022-04-19:
The nozzle is pressurized, therefore you need to do a retraction before travel (if there is some travel) to prevent oozing.

fontenele, 2022-05-21:
Yes but a lot of people does not have oozing, but have subextrusion because have a retract after M600 command.
I solved commenting retraction after M600 at gcode file, but it will be cool if has option to make or not the retraction.