prusa3d/PrusaSlicer #8149: Wipe tower requires a minimum of filament purge and this cannot be changed anywhere

mihai-stanimir, 2022-04-02:
### Description of the bug

Using multiple extruders with very low purging volumes: 0mm3 for unload and 5mm3 for load. Basically I'm aiming at a bit of priming before starting to print with a new hotend.

I expect the purge tower to only use the 5mm3 for each load as specified in the purging volumes.

What happens is that the purge tower uses up a minimum of 15mm3 of material on each load. This happens even if I set all purging volumes to zero (but still leave the tower on).

This 15mm3 of material will not move to infill when using "wipe into infill".

If there's a minimum purge amount, I'd like to be able to change it. Even if 15mm3 doesn't sound like much.
[image]

See the attached zip with a sample .3MF file

### Project file & How to reproduce

[minimum wipe tower issue.zip](https://github.com/prusa3d/PrusaSlicer/files/8401524/minimum.wipe.tower.issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+ win64

### Operating system

Windows 10

### Printer model

not a printer, but this happens with any multi extruder printer profile

rtyr, 2022-04-02:
There is a separate setting for minimal purge volume in filament settings, you can change it there (see screenshot)

[image]

mihai-stanimir, 2022-04-02:
Awesome, thank you @rtyr ! I didn't think of checking there 😁

closed 2022-04-02 by mihai-stanimir

justinds89, 2022-06-27:
@mihai-stanimir did you ever get this figured out? I tried adjusting the **Wipe Tower Parameters** under **Filament Settings** to reduce this, but it had no impact for me.

I really need to be able to have a very small wipe tower. It doesn't need to purge much. Just enough to get the filament flowing and wipe and globs off in the process before it starts the print.

mihai-stanimir, 2022-06-28:
@justinds89 yes, worked perfectly. Aside from changing the minimal purge, yYou have to also edit the purging volumes. It's a button on the right side on the main view, in between Supports and the list of STL files. Opens up a small window you can see in my original post. If you're still using a wipe tower, make sure there's enough wiping for the tower to have some structure and not collapse for being too skinny.

justinds89, 2022-06-28:
@mihai-stanimir Oh okay, I tried to adjust both of those. Even tried
setting it all at 0 to see what would happen. It seems like it would only
get to a certain amount purged and then stop regardless what was entered. I
will try again and maybe check what version of PrusaSlicer I'm on. Thanks
for the reply!

mihai-stanimir, 2022-06-28:
To be frank, I only checked the preview in the slicer and I didn't actually use the output g-code to print.