prusa3d/PrusaSlicer #8150: Crash with vase mode without G1 Z5 F5000 in start gcode

7ocb, 2022-04-02:
### Description of the bug

If custom start GCODE does not contain `G1 Z5 F5000 ; lift nozzle`, then slicer crashes trying to slice in vase mode:

```
(prusa-slicer:163690): Tracker-CRITICAL **: 18:57:08.173: tracker_sparql_connection_close: assertion 'TRACKER_IS_SPARQL_CONNECTION (connection)' failed

(prusa-slicer:163690): Gtk-CRITICAL **: 18:57:08.173: Could not init tracker3 search engine: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Tracker3.Miner.Files was not provided by any .service files

(prusa-slicer:163690): Tracker-CRITICAL **: 18:57:08.535: tracker_sparql_connection_close: assertion 'TRACKER_IS_SPARQL_CONNECTION (connection)' failed

(prusa-slicer:163690): Gtk-CRITICAL **: 18:57:08.535: Could not init tracker3 search engine: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Tracker3.Miner.Files was not provided by any .service files
/usr/include/c++/11.2.0/bits/stl_algo.h:3658: constexpr const _Tp& std::clamp(const _Tp&, const _Tp&, const _Tp&) [with _Tp = long unsigned int]: Assertion '!(__hi < __lo)' failed.
zsh: IOT instruction (core dumped)  prusa-slicer
```

### Project file & How to reproduce

[platers.zip](https://github.com/prusa3d/PrusaSlicer/files/8402893/platers.zip)

There are two plates, one is "working", which is working, and one is "crashing" which crashes slicer when trying to generate gcode.

The only difference is that "crashing" one does not have `G1 Z5 F5000 ; lift nozzle` in "printer settings -> custom gcode -> start gcode".

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Linux elkbook 5.16.15-arch1-1 #1 SMP PREEMPT Thu, 17 Mar 2022 00:30:09 +0000 x86_64 GNU/Linux

### Printer model

Ender 2

combolek, 2022-04-02:
Does it happen with the pre-built AppImage? I cannot reproduce it with `PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage` or the latest master built from source:
[image]

7ocb, 2022-04-03:
@combolek 

1. My fault! I just named files wrong :( 

Crashing one is `working`. There is updated zip

[platers-fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/8404011/platers-fixed.zip)
 
2. `AppImage` version does not crash, but instead seems to produce no traces:

[image]

combolek, 2022-04-03:
I can reproduce your result with the AppImage but I think this is only an issue with the preview. If you export the G-code file, it seems fine. You can even view reopen it in prusa-slicer to view it.

Also, this issue seems to be fixed in the latest master.

7ocb, 2022-04-03:
@combolek you are right. I built the 2.5.0-alpha0 and it seem to generate traces.

rtyr, 2022-04-04:
It is likely related to https://github.com/prusa3d/PrusaSlicer/issues/8032, which is already fixed in the master.

rtyr, 2022-04-05:
We were not able to reproduce the crash on Linux with 2.4.1. We will be releasing 2.4.2-rc soon, which will hopefully fix your issue. It would be nice, if you could retest it again with rc and let us know.

lukasmatena, 2022-04-05:
This should be fixed in the just released 2.4.2-rc1. Can you please confirm? Thanks.

7ocb, 2022-04-06:
@lukasmatena 

I tried `PrusaSlicer-2.4.2-rc1+linux-x64-GTK3-202204051358.AppImage` and `PrusaSlicer-2.4.2-rc1+linux-x64-GTK3-202204051358.tar.bz2 `, both seem to behave as expected: no crash, traces created and visible in preview.

rtyr, 2022-04-06:
Thanks. Closing.

closed 2022-04-06 by rtyr