prusa3d/PrusaSlicer #8151: PrusaSlicer is not opening/handleing Files, causes Error “Loading of a model file failed.”

RichardAlpert, 2022-04-02:
### Description of the bug

Since Mac OS 12.3 and later with Mac OS 12.3.1 (21E258) PrusaSlicer is not opening any File which worked before without issue. Regardless if using Drag'n Drop or via the Menu Path. 

Also it doesn't matter which Printer Profile you use and which Object File. STL and 3MF aint working. I already deleted PrusaSlicer including his hidden folders  (as far as I could found them all) but still no success) and re-installed it but the Problem is still there.

I already added a comment on this Report but I am not sure if it is still being monitored:
https://github.com/prusa3d/PrusaSlicer/issues/7987

### Project file & How to reproduce

PrusaSlicer
Version:   2.4.1+arm64
Build:     PrusaSlicer-2.4.1+arm64-202203101039

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Version 12.3.1 (Build 21E258)
Total RAM size [MB]: 17,180MB
OpenGL installation
GL version:   2.1 Metal - 76.3
Vendor:       Apple
Renderer:     Apple M1
GLSL version: 1.20
<details>
<summary>Installed extensions:</summary>
GL_APPLE_aux_depth_stencil
GL_APPLE_client_storage
GL_APPLE_element_array
GL_APPLE_fence
GL_APPLE_float_pixels
GL_APPLE_flush_buffer_range
GL_APPLE_flush_render
GL_APPLE_packed_pixels
GL_APPLE_pixel_buffer
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_specular_vector
GL_APPLE_texture_range
GL_APPLE_transform_hint
GL_APPLE_vertex_array_object
GL_APPLE_vertex_point_size
GL_APPLE_vertex_program_evaluators
GL_APPLE_ycbcr_422
GL_ARB_color_buffer_float
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_draw_buffers
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_instanced
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_instanced_arrays
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_provoking_vertex
GL_ARB_seamless_cube_map
GL_ARB_shader_objects
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shadow
GL_ARB_shadow_ambient
GL_ARB_sync
GL_ARB_texture_border_clamp
GL_ARB_texture_compression
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_non_power_of_two
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_transpose_matrix
GL_ARB_vertex_array_bgra
GL_ARB_vertex_blend
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_window_pos
GL_ATI_separate_stencil
GL_ATI_texture_env_combine3
GL_ATI_texture_float
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_clip_volume_hint
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_draw_buffers2
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_specular_color
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture_array
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_s3tc
GL_EXT_texture_env_add
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod_bias
GL_EXT_texture_rectangle
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_timer_query
GL_EXT_transform_feedback
GL_EXT_vertex_array_bgra
GL_IBM_rasterpos_clip
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_light_max_exponent
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SGI_color_matrix
</details>

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1 +arm64

### Operating system

Mac OS 12.3.1

### Printer model

Prusa Mini; Prusa MK3s

RichardAlpert, 2022-04-02:
[image]

RichardAlpert, 2022-04-03:
I made now a downgrade to PrusaSlicer 2.3.3 and it ain't working.

lukasmatena, 2022-04-04:
@RichardAlpert Thanks. Could you please run PrusaSlicer from the command line as `prusa-slicer --loglevel=5` and paste its output after the failed import? There might be some more info in the log.

RichardAlpert, 2022-04-04:
> @RichardAlpert Thanks. Could you please run PrusaSlicer from the command line as `prusa-slicer --loglevel=5` and paste its output after the failed import? There might be some more info in the log.

Hello,
please tell me how to do on Apple OS, M1 Mac. I am not familiar with this method, so please provide me the instructions.

lukasmatena, 2022-04-04:
I'll try. You download the DMG from our github, open it and drag and drop the application somewhere (e.g. to the desktop). Right click the application, select "Show package contents". A finder window will appear, change directory into "Contents/MacOS". You'll see a file called PrusaSlicer. Now open a blank terminal window next to it and drag and drop the PrusaSlicer file in there. It will type the path of the file. Append `--loglevel=5` and confirm. PrusaSlicer will start and you will do what you were doing before. When the import fails, see the terminal window (now full of tracing output from PrusaSlicer) and copy-paste the output here.
Thanks.
[image]

RichardAlpert, 2022-04-04:
Thank you, worked like you told. 

Last login: Mon Apr  4 12:34:10 on ttys000
myname@MacBook-Pro-von-NAME ~ % /Applications/Original\ Prusa\ Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer --loglevel=5
[2022-04-04 14:14:39.956918] [0x0000000107450580] [trace]   Initializing StaticPrintConfigs
[2022-04-04 14:14:39.969273] [0x0000000107450580] [debug]   full path: 7484420837251052083
[2022-04-04 14:14:39.969287] [0x0000000107450580] [debug]   single instance: undefined. other params: "/Applications/Original Prusa Drivers/PrusaSlicer.app/Contents/MacOS/PrusaSlicer";--loglevel=5
[2022-04-04 14:14:39.969291] [0x0000000107450580] [debug]   full lock path: /Users/NAME/Library/Application Support/PrusaSlicer/cache/7484420837251052083.lock
[2022-04-04 14:14:39.969401] [0x0000000107450580] [debug]   Creating lockfile.
[2022-04-04 14:14:39.969406] [0x0000000107450580] [info]    Instance check: Another instance not found or single-instance not set.
[2022-04-04 14:14:40.067168] [0x0000000107450580] [trace]   System language detected (user locales and such): de_DE
[2022-04-04 14:14:40.067787] [0x0000000107450580] [trace]   Best translation language detected (may be different from user locales): de_DE
[2022-04-04 14:14:40.067795] [0x0000000107450580] [trace]   Switching wxLocales to de_DE
[2022-04-04 14:14:40.110348] [0x0000000107450580] [info]    Checking if indices need to be installed from resources...
[2022-04-04 14:14:40.268655] [0x0000000107450580] [info]    3dx mac handler starts
[2022-04-04 14:14:40.268704] [0x0000000107450580] [error]   3dx drivers module loading error: dlopen(/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient, 0x0005): tried: '/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file), '/System/Library/Frameworks/3DconnexionClient.framework/3DconnexionClient' (no such file)
[2022-04-04 14:14:40.268708] [0x0000000107450580] [info]    3dx drivers loaded: NO
[2022-04-04 14:14:40.269952] [0x0000000107450580] [info]    Undo / Redo snapshot taken: Neues Projekt, Undo / Redo stack memory: 0MB Resident memory: 115MB; Peak memory usage: 115MB
[2022-04-04 14:14:40.297907] [0x0000000107450580] [debug]   BuildVolume bed_shape clasified as: Rectangle
[2022-04-04 14:14:40.316792] [0x0000000107450580] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2022-04-04 14:14:40.316803] [0x0000000107450580] [debug]   GLCanvas3D: Use Retina OpenGL: true
[2022-04-04 14:14:40.440133] [0x0000000107450580] [debug]   TriangleMesh::repair() started
[2022-04-04 14:14:40.440431] [0x0000000107450580] [debug]   TriangleMesh::repair() finished
[2022-04-04 14:14:40.902773] [0x0000000107450580] [info]    Checking for cached configuration updates...
[2022-04-04 14:14:40.902794] [0x0000000107450580] [info]    Confing bundle not installed for vendor TriLAB, skipping: 
[2022-04-04 14:14:40.902798] [0x0000000107450580] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2022-04-04 14:14:40.902801] [0x0000000107450580] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2022-04-04 14:14:40.909050] [0x0000000107450580] [debug]   Vendor: Prusa Research, version installed: 1.4.4, version cached: 1.4.4
[2022-04-04 14:14:40.909063] [0x0000000107450580] [info]    Confing bundle not installed for vendor Ultimaker, skipping: 
[2022-04-04 14:14:40.909067] [0x0000000107450580] [info]    Confing bundle not installed for vendor Voron, skipping: 
[2022-04-04 14:14:40.909070] [0x0000000107450580] [info]    Confing bundle not installed for vendor Jubilee, skipping: 
[2022-04-04 14:14:40.909073] [0x0000000107450580] [info]    Confing bundle not installed for vendor FLSun, skipping: 
[2022-04-04 14:14:40.909076] [0x0000000107450580] [info]    Confing bundle not installed for vendor Anycubic, skipping: 
[2022-04-04 14:14:40.909079] [0x0000000107450580] [info]    Confing bundle not installed for vendor gCreate, skipping: 
[2022-04-04 14:14:40.909082] [0x0000000107450580] [info]    Confing bundle not installed for vendor Creality, skipping: 
[2022-04-04 14:14:40.909085] [0x0000000107450580] [info]    Confing bundle not installed for vendor RatRig, skipping: 
[2022-04-04 14:14:40.909139] [0x0000000107450580] [info]    Confing bundle not installed for vendor Artillery, skipping: 
[2022-04-04 14:14:40.909146] [0x0000000107450580] [info]    Confing bundle not installed for vendor INAT, skipping: 
[2022-04-04 14:14:40.909148] [0x0000000107450580] [info]    No configuration updates available.
[2022-04-04 14:14:40.963444] [0x000000017b5cf000] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2`
*   Trying 185.115.1.124:443...
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* TLS 1.2 connection using TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384
* Server certificate: files.prusa3d.com
* Server certificate: R3
* Server certificate: ISRG Root X1
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version2 HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.4.1+arm64
Accept: */*

* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 04 Apr 2022 12:14:42 GMT
< Content-Length: 6
< Connection: keep-alive
< Last-Modified: Mon, 04 Apr 2022 12:00:35 GMT
< ETag: "6-5dbd2e03dc01f"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[... cut ...]
[2022-04-04 14:14:42.652590] [0x000000017b5cf000] [warning] No such vendor: Artillery
[2022-04-04 14:14:42.652610] [0x000000017b5cf000] [warning] No such vendor: INAT
[2022-04-04 14:15:07.134846] [0x0000000107450580] [info]    Undo / Redo snapshot taken: Datei laden: interno-piccolo-v2.stl, Undo / Redo stack memory: 0MB Resident memory: 629MB; Peak memory usage: 970MB
[2022-04-04 14:16:01.664549] [0x0000000107450580] [info]    Undo / Redo snapshot taken: Datei laden: star-split-bottom-black.stl, Undo / Redo stack memory: 0MB Resident memory: 604MB; Peak memory usage: 970MB

RichardAlpert, 2022-04-04:
I tried it with two different files, both .stl.

RichardAlpert, 2022-04-05:
Yes, correct. BUT! It is working with .3MF and .STL Files which came with the Prusa Installation Package. Other files which worked flawless so far are not working anymore. I have no clue, it makes no sense to me.

lukasmatena, 2022-04-07:
@RichardAlpert So far we were unable to reproduce the issue. Could you please post one or two (ideally simple) STL files so we can try on the same data? We tried on the one provided by @DominikHoffmann in #7987, but we found no problem.

The log that you provided did not help in finding the cause. If you really posted the log after you tried the import, it means that there is no logging in the defective path. I created a special build based on 2.4.1-rc1 with extra tracing (download link below). Could you please try it, check whether the problem is still there and if so, create the log the same way as before? Thanks.

Link to download the testing build: [PrusaSlicer-2.4.2-rc1+1-lm-macos-model-loading-universal-g19e90ba61](https://drive.google.com/file/d/1-mYbrffuetKhfDd5MEi7KDyGiFZC3_cI/view?usp=sharing)

RichardAlpert, 2022-04-07:
Hello,
I did the log like you told me. I will uninstall the actual PrusaSlicer and will use your "Special Edition" Tool but unfortunately I am not able to acces your file due to missing permissions. I asked for access (my mail is de.**.**).

In the meantime here are some files which worked 100% before the issue came up.
[TEST STL_Files.zip](https://github.com/prusa3d/PrusaSlicer/files/8442557/TEST.STL_Files.zip)

jmd73, 2022-04-07:
Hello!

@RichardAlpert I tested your files (on Windows 10) and they seem to be filled with zeroes only. (unzipped with 7Zip and Windows built in zip).

This is the beginning of one of the files in an hex editor:
[image]

I get the same result if I open the files in for example notepad++.

Can you check the files in an text editor or a hex editor?

Best regards, Juha D.

lukasmatena, 2022-04-07:
@RichardAlpert My mistake, I have now lifted permissions so anyone can download (hopefully).

But I also agree with what @jmd73 has said. These files seem to contain no useful data, the error message is quite appropriate.

The file provided in #7987 is not visibly broken however, and I was able to load it.

lukasmatena, 2022-04-07:
Where did you get those files? Did you create them yourself? Maybe it is the generator that is broken on macOS 12.x.x, not our beloved slicer.

RichardAlpert, 2022-04-07:
Dear all,
I am shocked and completely speechless. The hint that the files would not contain any information made me take a closer look at the files and I could not believe it. It is as described by you, no content there.

This applies to all the files I have. I can't understand what happened here, the files are stored on OneDrive, without encryption or other changes and were simply dragged from the desktop or linked folder into the slicer.

Why now the files from the shortcut are empty, I do not understand. If I download the files directly from OneDrive and then drag them into the slicer, everything works as it should. Also, the "empty" files have a file size, while "real" empty files should have almost no file size. I suspect that MacOS misinterpreted the files and changed something in the background.

By the way, I see now that intact STL files on MacOS always have a small preview image as thumbnail, which the empty STL files do not have.

Of course the error message makes sense then. I feel pretty stupid now to have bothered you with this, I am infinitely sorry to have wasted your time. But maybe you can redesign the error message so that in such cases it gives a feedback like "The file contains no data" or "The file seems to be corrupted". 

I am now trying to determine if the error is coming from OneDrive or from the update of macOS. 

Which makes me wonder a bit now though. I had also sent files to the support in the chat and they said everything was OK, they could not trace the error. How can this be then? Well, this should not bother you now.

Sorry again for the unnecessary stress.

bubnikv, 2022-04-07:
Thanks for heads up. It is scary, that a similar issue has been reported by multiple users.

bubnikv, 2022-10-11:
Would you please attach those files? You need to zip them first.