prusa3d/PrusaSlicer #8156: Automatic color change for logo crashes PrusaSlicer with second object of 0.2mm height

fe60, 2022-04-04:
### Description of the bug

My object has been detected potentially being a logo. Adding the proposed automatic color change crashes the application.
This seems to be related to the simple box I have added as custom brim.

The example below shows a simplified model, but still allows to reproduce the crash.
Clicking the action/link in the notification crashes the application.
[image]

### Project file & How to reproduce

1. Open attached project
2. Slice
3. Apply proposed automatic color change
=>Crash

[PrusaCrashAutomaticLogo.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8408054/PrusaCrashAutomaticLogo.3mf.zip) (just remove file ending .zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 10, 21H2

### Printer model

Prusa i3 MK3S+

FidelCapo, 2022-04-05:
This issue should be fixed and fix will be part of 2.3.2 release.
Closing.

FidelCapo, 2022-04-11:
Thank you for reporting. This bug has been fixed and will be part of the next release.
Closing.