prusa3d/PrusaSlicer #8157: Can't set the Z position on an STL file.

traderhut, 2022-04-04:
### Description of the bug

Two color print.  I have a coaster that is 5mm high.

Extruded down from the face of it is an image (doesn't matter what) that is extruded down twice,  both times 0.4mm.  The first time it is a cut, the 2nd time it is a 'new body'

This creates a cut out in the coaster for the image that would basically be the same as a 'inlay' in wood.
And the inlay as a different body.
I export both bodies as STL files.

When I drop either of them on the plater, I get their position as Z= 0.2mm  Which, I can't change.  I need the 2nd stl (inlay) to be at 4.6mm for the z, so when it is sliced it will switch colors ONLY at the top of the print, for the last 2mm, keeping the same color for the entire bottom 4.6mm of the coaster.

I don't understand:
   1. Where it is getting 0.2mm for the Z  (Layer height??)
   2. Why I can't change it to move it up in the Z axis.
   3. What World/Local coordinates means

I remember doing something like this with RepetierHost, but I don't recall if I ever did it without it going all the way down or not.

And not sure if this should be a feature request instead.

### Project file & How to reproduce

Drop two STL files down.

Attempt to change the Z for either.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5 Alpha (built from current source)

### Operating system

Windows 10

### Printer model

E3D Tool changer, but doesn't matter

neophyl, 2022-04-04:
Duplicate of one of the longest most contentious posts here https://github.com/prusa3d/PrusaSlicer/issues/1513.

lukasmatena, 2022-04-05:
Duplicate of #1513. Closing.

closed 2022-04-05 by lukasmatena