prusa3d/PrusaSlicer #8160: 2.4.1 freezes when selecting a .3mf file

tony-ozrelic, 2022-04-04:
### Description of the bug

The app freezes when I click on a .3mf file.

If I launch another copy, clicking on a .stl file works fine. 

Multiple clicks gray out the file dialog and change the dialog banner to 'Not Responding'.

I can kill the frozen app with Task Manager. 

If I use File->Recent, the .3mf file loads properly. 

As I was trying to drag-n-drop config.ini to this report, I noticed that Windows Explorer was freezing when I click on a .3mf file as well!

Turning off Preview in Explorer keeps it from freezing. 

Something about Preview in Explorer causes clicking on a .3mf file to hang Explorer as well as the File Selection dialog in 2.4.1.

### Project file & How to reproduce

[config.zip](https://github.com/prusa3d/PrusaSlicer/files/8413678/config.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Device name	tonys-desktop Processor	Intel(R) Core(TM) i7-8809G CPU @ 3.10GHz   3.10 GHz Installed RAM	16.0 GB Device ID	B5A6A02F-B7EE-47DB-B28F-AAE70B5218CF Product ID	00330-80013-78270-AA066 System type	64-bit operating system, x64-based processor Pen and touch	Pen support

### Printer model

MK3S

neophyl, 2022-04-05:
Sounds like you just reported a bug in windows explorer.

lukasmatena, 2022-04-05:
@tony-ozrelic Thanks, but to be honest, I'm not sure I understand exactly.

> The app freezes when I click on a .3mf file.

Which app? Explorer or PrusaSlicer? Is PrusaSlicer running at this point? If not, does it start?

> If I launch another copy

Another copy of what?

> Multiple clicks gray out the file dialog and change the dialog banner to 'Not Responding'.

Clicks on what? Where? Is it the Explorer dialog or Open File in PrusaSlicer?

Could you maybe record a short video so it is clear what we are talking about? What is your exact operating system version?

tony-ozrelic, 2022-04-05:
> @tony-ozrelic <https://github.com/tony-ozrelic> Thanks, but to be honest,
> I'm not sure I understand exactly.
>
> The app freezes when I click on a .3mf file.
>
> Which app? Explorer or PrusaSlicer? Is PrusaSlicer running at this point?
> If not, does it start?
My apologies, here are the steps:

- launch PrusaSlicer
- File->Open opens a file selection dialog
- click on a .3mf file. Nothing happens. Continue clicking. The dialog is
grayed out and the dialog's banner changes to 'Not Responding'

> If I launch another copy
>
> Another copy of what?
PrusaSlicer

> Multiple clicks gray out the file dialog and change the dialog banner to
> 'Not Responding'.
>
> Clicks on what? Where? Is it the Explorer dialog or Open File in
> PrusaSlicer?
see steps above.

> Could you maybe record a short video so it is clear what we are talking
> about? What is your exact operating system version?
Sorry, here is the build info:

Edition Windows 10 Pro
Version 21H2
Installed on ‎12/‎11/‎2020
OS build 19044.1586
Experience Windows Feature Experience Pack 120.2212.4170.0

I suspect that the Open dialog is calling something that reads the .3mf
file I clicked on and generates or renders something to put in the Preview
subwindow of the dialog, but this task is hung for some reason and the
dialog is tagged by the OS as 'Not Responding'.

Similar behavior is seen when clicking on a .3mf file in Windows Explorer.

It appears I have stumbled upon something that most systems handle without
a second thought.

Perhaps I am missing some part of the OS that completes the task of
previewing a .3mf file?

Anyway, my workaround is to turn off the preview window or just not click
on a .3mf file for now.

lukasmatena, 2022-04-05:
Thanks.

> Similar behavior is seen when clicking on a .3mf file in Windows Explorer.

If you have problems with the Explorer even when PrusaSlicer is not running, it is quite likely that the problem is somewhere else. PrusaSlicer itself does not do anything when you work with the Open File dialog, that is a system dialog that we ask the system to show, we have little control over what it does. It really sounds as if there was some problem with the preview.

Can you post a 3MF that you know creates this problem? It is also possible that there is something wrong in the file itself (although probably not likely).

bubnikv, 2022-04-05:
Did you install any Explorer extension that renders 3MFs? Did you install any other slicer, for example
https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file
?

tony-ozrelic, 2022-04-05:
if I look at Windows Settings->Choose default apps by file type, the .3mf
file extension launch app is PrusaSlicer.

The default for .stl files is Marlin3DPrinterSTLViewer.

If I change the default viewer for .stl files to PrusaSlicer, the issue
goes away - clicking on a .stl or .3mf file now launches PrusaSlicer, and
after a slight pause, the Windows Explorer preview window is rendered
properly.

launching PrusaSlicer and clicking File->Open Project now launches the File
Open dialog, and clicking on a .3mf now previews properly.

So, it looks like some sort of issue with the .stl and the .3mf extensions
launching different apps?

Anyway, so far as I am concerned, this issue can be closed now.

thanks,

tony o.

Ventusfahrer, 2022-04-08:
Hi, 

I have the same problem with the 2.4.1-Linux-AppVersion on Ubuntu 21.10 (with current updates) and on second computer with Manjaro Linux (latest updates installed).

File I try to load: https://media.printables.com/media/prints/3673/stls/158818_1b8490e9-887d-4b11-b219-0c68d3e0d026/prusa_mmu_enclosure_11_black.3mf#_ga=2.154099704.1511522116.1649245004-2020314068.1634560730

In the bash-shell I started the Slicer in, I get the error message:
xxxx:~/bin$ ./PrusaSlicer-2.4.1+linux-x64-GTK3-202203101056.AppImage 
[2022-04-08 13:42:55.187252] [0x00007fecf302b8c0] [trace]   Initializing StaticPrintConfigs
free(): invalid pointer
Aborted (core dumped)

Do you need additional information?

Thanks for supporting!

regards

Peter

lukasmatena, 2022-04-08:
@Ventusfahrer I highly doubt that this is the same issue as @tony-ozrelic has. It sound more like a duplicate of #8078, which is already fixed in [2.4.2-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc1). Can you confirm that?

lukasmatena, 2022-04-08:
As for the original issue:

> Anyway, so far as I am concerned, this issue can be closed now.

I agree. Apparently not a bug in PrusaSlicer. Closing.

closed 2022-04-08 by lukasmatena

Ventusfahrer, 2022-04-08:
> @Ventusfahrer I highly doubt that this is the same issue as @tony-ozrelic has. It sound more like a duplicate of #8078, which is already fixed in [2.4.2-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc1). Can you confirm that?

1st Confirmation: Yes - you are right, I posted the problem into the wrong issue
2nd Confirmation: 2.4.2.-rc1 is fixing the issue (just test it)

Thanx for replying!

kind regards

Peter