prusa3d/PrusaSlicer #8164: Slicing slows or hangs on "Generating Permiters 20%" cpu load is at 100%

mattfox27, 2022-04-05:
### Description of the bug

When slicing complex models Prusaslicer seems to hang or even crash on generating perimeter's 20% also cpu load goes to 100%

### Project file & How to reproduce

Slice a complex model

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

win 10

### Printer model

prusa mk3+

rtyr, 2022-04-06:
Please provide your 3MF project file.

lukasmatena, 2022-04-12:
Thanks. It is caused by `Print Settings->Advanced->Fill gaps`. The gap fill algorithm has a hard time processing the model. We should improve it.

boelle, 2022-09-08:
same happens in 2.5.0

wanted to print a simple lamp shade and it also hangs for a very long time at 20% and uses 100% cpu
[kitchen lamp shade.zip](https://github.com/prusa3d/PrusaSlicer/files/9527361/kitchen.lamp.shade.zip)

chrisl8, 2022-09-17:
Wow, ran into this same thing with a chain I've printed many times before. It used to work fine, but now it hangs at 20% where is spins up all 24 cores of my CPU to 100% forever! (Well, I only wanted 10 or 15 minutes, but really that should be enough).

https://www.dropbox.com/s/g4h1ns75bituwbg/420_Chain_84_link.stl?dl=1

Turning off Fill Gaps does not help.

Area5142, 2022-09-17:
> [image]

If I use _Arachne Perimeter Generator_ or _Classic Perimeter Generator without Gap fill_, the slice time is now about 10s, which is the same as Cura uses.

However, if I use the _Classic Perimeter Generator with Gap fill_, the slice time is still long (about 6 min), for this relatively simple model.

shwekwek, 2022-10-21:
Same exact issue with the Nasa chainmail file on printable site.
Up to 8X8 works fine, above that it simply get stuck.
https://www.printables.com/model/205674-nasa-chainmail-4x4-8x8-12x16-and-32x34/files
I couldnt find anything to change on the settings that would fix that

JayKeyStk, 2022-10-23:
file: https://media.printables.com/media/prints/205674/stls/1890102_5d284bd2-ccf9-48b0-8f30-4bbb62cd30ce/nasa_fabric_mk_3_thick_32x24_lo_tnt.stl#_ga=2.19193535.689778600.1666509640-2016993698.1648644103

Freezes at 20% after scaling down to fit on mk3 bed. 
CPU load varies between 75-100%
2.5.0

:-/

foreachthing, 2022-10-23:
> Freezes at 20% after scaling down to fit on mk3 bed. CPU load varies between 75-100% 2.5.0

It did not freeze on my computer - it just took a very, very time long to slice.

Area5142, 2022-10-23:
On Linux Mint 19.3 PrusaSlicer 2.5.0 the "Nasa chainmail" file freezes my computer after 5-10 min - hard boot is needed...

bubnikv, 2022-11-15:
I did some optimizations for upcoming PrusaSlicer 2.6.0-alpha.

> wanted to print a simple lamp shade and it also hangs for a very long time at 20% and uses 100% cpu
That model seems to slice fine now, I am not sure how fast it sliced before.

> NASA_Fabric_Mk_3_Thick_32x24_LO_TnT.stl
This is one crazy model. With my optimizations for 2.6.0-alpha it slices about half an hour on my 12-core desktop.

> 420_Chain_84_link.stl
It slices fine now.

> https://www.printables.com/model/81180-primer-flipping-tray
The gap fill algorithm with "classic" perimeter generator still runs very slowly. I may look into it.

bubnikv, 2022-11-16:
> https://www.printables.com/model/81180-primer-flipping-tray
> The gap fill algorithm with "classic" perimeter generator still runs very slowly. I may look into it.

The gap fill slowness will be fixed in PrusaSlicer 2.6.0-alpha1.
Closing.

closed 2022-11-16 by bubnikv