prusa3d/PrusaSlicer #8165: White screen and dots

krisdusz, 2022-04-05:
### Description of the bug

PrusaSlices stopped displaying menu and bed texture.
I did reinstall and removed my settings but problem reoccurred just after skiping initial settings.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Wersja 2.4.1+win64

### Operating system

Windows11Pro

### Printer model

Ender 3 v2 with Astrobox

lukasmatena, 2022-04-11:
Could you please share your GPU information? We were not able to reproduce the issue, although there is a similar report already (#8180). We are planning some changes in how OpenGL is handled in PrusaSlicer, we'll see if they help.

krisdusz, 2022-04-12:
MSI 3080 Trio Gaming Z

lukasmatena, 2022-04-12:
@krisdusz Thanks. Author of #8180 (@peterhieuvu) reported MSI Ventus 3x 3070. It can also be a driver issue. It might be a good idea to keep it updated.

lukasmatena, 2023-03-27:
@krisdusz Is this still happening with 2.6.0-alpha5? Thanks.

lukasmatena, 2023-03-28:
Thanks. Closing.

closed 2023-03-28 by lukasmatena