prusa3d/PrusaSlicer #8166: PrusaSlicer iOS/iPadOS app(s)

TeslaCT, 2022-04-05:
Working with PrusaSlicer requires a machine running a desktop environment. Remote access programs are an option but can be cumbersome as they are not optimized for mobile devices like phones and tablets. 

The community would like to see dedicated PrusaSlicer apps for iOS and iPadOS. 

Using an app on a mobile device would be great for adjusting slicing settings, viewing previews, and generating G-code. Then we could send via Octoprint to our connected printers.

edelmaca, 2022-08-27:
An iPadOS Version would be really great. Designing with Shapr3D and Slicing with PrusaSlicer for iPad. That would make my MacBook obsolete. That's really the only task my old MacBook has to handle these days.
Im no Developer by any means. But Josef Prusa mentioned a iPad version at some point in a prusa live show, that it could be doable because they develop a ARM Version for the Mx MacBooks and the iPad Pro Models also house an M Processor. That sounds logic to me. But I don't know what one has to do to port it from MacOS to iPadOS/iOS. I could imagine the hurdles now are much smaller compared to the A Processor times.

I would love to use PrusaSlicer on my iPad Pro.

grassy-666, 2022-09-06:
I am still wanting this.

dorkness-io, 2022-10-21:
A big +1 to this feature. Using Shapr3D + Prusaslicer on the iPad would (with Apple Pencil support) would be a great combination on the iPad Pro.

Delvien, 2022-12-11:
I too hope for this! Prusaslicer is already built for arm64 so half the work is done (im no dev but I would think this was the case!)

jdnichollsc, 2022-12-28:
OMG, the community needs this! <3

johnkray, 2023-01-11:
I would really like to see this as well! This would be very useful for me.

DavidBjerreBjoerklund, 2023-02-12:
I can only agree! This would be extremely helpful and Ipads should be powerful enough by now.

Korrrrrrl, 2023-03-06:
This would be so helpful. I am also waiting for the slicer to come to ipad.

Zaaphod42, 2023-08-05:
Yes, please, do that! :) 
I would love to be able to check for printing progress on an app or, ideally, an iOS widget.

william-family, 2023-08-20:
It seems to me that excuses of being a community driven company does not work anymore. Prusa is currently, let’s be honest, a behemoth in 3d printing which leads the way. Having an iPadOS app is now a REQUIREMENT to survive. Well, unless you want to be a Kodak in a digital photography era. 
The hardware resources of iPad’s are not just enough. They exceed many PC’s used by the enthusiasts building and slicing models. 
Grow up and release an iPadOS app finally!

Madad1996, 2023-12-09:
Wouldn't be a huge task to get it running on an iPad given that wxWidgets is running on iOS too, maybe someone could get it running relativly quick

darkmastermindz, 2024-03-02:
Very interested in this as well

werdnum, 2024-03-05:
I took a look out of curiosity. PrusaSlicer uses the wxWidgets library, which has a [rudimentary iOS implementation](https://docs.wxwidgets.org/latest/plat_ios_install.html). I am assuming that if we wanted a full-scale PrusaSlicer implementation on iOS, you'd probably need to improve the wxiOS library first.

dhyananbho, 2024-05-21:
There are various CAD programs for the iPad (Autodesk Fusion, Shapr3D, Onshape, Tinkercad, ...). It would be really nice if PrusaSlicer was also available on iPadOS so that you can do the whole 3D printing process on the iPad. In addition, Apple has just released new iPads, where the Pro model even has the most powerful M4 processor inside - definitely plenty of computing power for PrusaSlicer. Please make PrusaSlicer available as an iPadOS app in the AppStore soon. Thank you very much.

AssetBurned, 2024-09-24:
Putting an +1 in here too.

dhyananbho, 2026-09-02:
There is still no PrusaSlicer available on iPadOS, and there are still people who would like to see PrusaSlicer natively supported on iPadOS.

lukasmatena, 2026-09-10:
I am moving this to Discussions as a feature request, in line with our new GitHub policy.

You may or may not have seen [this recent Twitter post](https://x.com/josefprusa/status/2097008626703778056) from Josef Prusa. It is not a promise of anything, just a hint that this was not forgotten in the 3.x rewrite.