prusa3d/PrusaSlicer #8168: Simple STL file crashes PrusaSlicer

anna328p, 2022-04-05:
### Description of the bug

I have an STL file of a sphere modified for printing.

[image]

Attempting to switch print profiles while 5% infill is selected results in PrusaSlicer crashing.

### Project file & How to reproduce

[ball-prusaslicer-crash.zip](https://github.com/prusa3d/PrusaSlicer/files/8422264/ball-prusaslicer-crash.zip)

The following sequence of steps crashes PrusaSlicer:

1. Import the STL file
2. Use "Place on Face" to rotate the file with the flat, round face down:
[image]
3. Select the "0.25mm DRAFT", "Prusament PLA", "Original Prusa MINI and MINI+" settings:
[image]
4. Select 5% infill.
5. Try to switch to a different print settings profile.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.1

### Operating system

Linux 5.17.1, NixOS, 22.05 (Quokka), 22.05.20220330.1063244

### Printer model

Prusa MINI+

anna328p, 2022-04-05:
Video: https://user-images.githubusercontent.com/9790772/161862802-a8097fcf-accb-4673-b63d-d800e729c926.mp4

lukasmatena, 2022-04-08:
@anna328p Thanks for the report, but we were not able to reproduce it. Would you please try to collect some data so we see where it crashes? It might be enough for us to realize what the problem is.

Generating a backtrace using `gdb` would be a good start. In short:
1. Download binary of the latest 2.4.1-rc1 release (https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.2-rc1/PrusaSlicer-2.4.2-rc1+linux-x64-GTK3-202204051358.tar.bz2) and unpack it
2. Open terminal window, cd to the folder where you unpacked the archive and into `bin/` subfolder
3. Type `gdb prusa-slicer` and confirm, wait until a prompt saying "(gdb)" appears
4. Type `run` and confirm (PrusaSlicer will start)
5. Reproduce the problem and let PrusaSlicer freeze
6. Go back to the terminal window. if you see `--Type <RET> for more, q to quit, c to continue without paging--`, press Enter until you get the (gdb) prompt again
7. Type `bt` and confirm (backtrace will be printed)
8. Type `quit` and confirm. You'll be asked if you really, want to quit - `y`, confirm.
9. Then just copy everything from the console and paste it here.

If you get some "command gdb not found" at step 3, install `gdb` package from your package manager first.

Thanks.

anna328p, 2022-04-08:
Can't reproduce on 2.4.1-rc1. Might be a problem with the settings transfer dialog?

lukasmatena, 2022-04-09:
@anna328p I made a mistake in the above post, I meant to write 2.4.2-rc1 in step 1. The URL is correct, though. Please, make sure you checked in 2.4.2-rc1. (Btw. 2.4.1-rc1 is pretty much the same as 2.4.1.) Sorry, my fault.

If you cannot reproduce it in 2.4.2-rc1, it means, that it was either fixed by some seemingly unrelated change (which is IMHO unlikely, not that much has changed between 2.4.1 and 2.4.2-rc1) or it does not happen every time. Anyway, if you manage to provide the backtrace from ANY version where you can reproduce it, it would be helpful.

> Might be a problem with the settings transfer dialog on my version?

There really is an issue (#7905) with that dialog that we recently fixed. The fix is not released yet, but that problem did not show when running in Expert mode. You should not see it (according to the video, you are running in Expert mode). So I doubt yours is the same.

anna328p, 2022-04-10:
[prusaslicer-backtrace.log](https://github.com/prusa3d/PrusaSlicer/files/8459822/prusaslicer-backtrace.log)
[prusaslicer.log](https://github.com/prusa3d/PrusaSlicer/files/8459823/prusaslicer.log)

lukasmatena, 2022-04-11:
@anna328p Thanks. The backtrace seems to indicate that it dynamically links to `/nix/store/xy8jjia97n2h1mjs4yfikw3hfx3y1aav-wxwidgets-3.1.5/lib/libwx_gtk3u_core-3.1.so.5`. Builds from our build server (such as [this one](https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.4.2-rc1/PrusaSlicer-2.4.2-rc1+linux-x64-GTK3-202204051358.tar.bz2)) link wxWidgets statically so I assume that what you are running is not in fact our build.

Our builds are linked against a modified version of wxWidgets 3.1.4. You link to 3.1.5 which we never tested. It looks that the bug is in fact a bug in wxWidgets 3.1.5: https://github.com/wxWidgets/wxWidgets/issues/19288.

I doubt there is anything we can do.