prusa3d/PrusaSlicer #8170: Right clicking on a STL and selecting "Open With PrusaSlicer .app" does not work correctly.

slwise, 2022-04-06:
### Description of the bug

If I right click on a STL then select "Open With PrusaSlicer.app" on my MacBook Pro it launches PrusaSlicer but the model does not show on the plater. It is listed on the right hand side but also has no dimensions listed. You can click the slice button then everything appears and ready to export the g-code. If I open PrusaSlicer, then open the STL everything works correctly. It does this with any STL I have tried. I have included a screen shot to show what is displayed when using "Open With".

[image]

### Project file & How to reproduce

[Scientist_Test_90mm.stl.zip](https://github.com/prusa3d/PrusaSlicer/files/8433279/Scientist_Test_90mm.stl.zip)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2-rc1

### Operating system

macOS Monterey 12.3.1

### Printer model

Ender 3 Pro with Micro Swiss Hotend

JohnOCFII, 2022-04-07:
I was just about to post the same thing.  I've got a 3MF generated from Fusion 360 that exhibits the same behavior.  Behavior is as expected in 2.4.1, but item does not show on the plater in 2.4.2-rc1 unless I hit the `Arrange` button, or slice the model.  Also on a Mac - 2021 MacBook Pro with M1 Pro.  Running Monterey
[image]
<img width="1582" alt="Screen Shot 2022-04-06 at 10 3
[Tea Kettle Base.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8439436/Tea.Kettle.Base.3mf.zip)
6 47 PM" src="https://user-images.githubusercontent.com/7844212/162115798-39624e6f-a5a3-465d-8843-0c5862798d94.png">
 12.3.1.

JohnOCFII, 2022-04-07:
Here's the 3MF
[Tea Kettle Base.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8439442/Tea.Kettle.Base.3mf.zip)

bubnikv, 2022-04-07:
Thanks for heads up. We are investigating the issue. It seems the issue is new with the update of our build system and it only affects the newer OSX versions.

bubnikv, 2022-04-07:
Hopefully fixed with https://github.com/prusa3d/PrusaSlicer/commit/6a67a6e21bee3bc99144d13cc760ccea50800e9c on stable.
Loading of files is postponed until the application is fully initialized.

slwise, 2022-04-13:
I just downloaded rc2 and the problem is fixed with "Open With" and Fusion 360. Thanks!!

JohnOCFII, 2022-04-13:
> You can check the fix in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). Closing the issue.

Can confirm it works for me as well. 👍