prusa3d/PrusaSlicer #8171: Simplified Chinese localization problems with some characters

xiaojie0815, 2022-04-07:
### Description of the bug

Some Chinese characters are directly turned into question marks, such as:辑、廓
[image]

[image]

This warning message also cannot be localized in Chinese
[image]

### Project file & How to reproduce

Not required

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

版本 2.4.2-rc1+win64

### Operating system

win10

### Printer model

voron

bubnikv, 2022-04-07:
All the Chinese characters to be rendered in the 3D scene need to be listed
out and registered in
https://github.com/prusa3d/PrusaSlicer/blob/a4435d214f3c5ed4111e474666fde5d431ea3501/src/slic3r/GUI/ImGuiWrapper.cpp#L182

We need to add your characters or character ranges here:
https://github.com/prusa3d/PrusaSlicer/blob/a4435d214f3c5ed4111e474666fde5d431ea3501/src/imgui/imgui_draw.cpp#L2861

Ideally the character ranges would be extended at the upstream ImGUI project
ocornut/imgui: Dear ImGui: Bloat-free Graphical User interface for C++ with
minimal dependencies (github.com) <https://github.com/ocornut/imgui>

xiaojie0815, 2022-04-07:
@bubnikv 
Thank you very much for your answer, I understand. I will revise and resubmit the code

closed 2022-04-07 by xiaojie0815

xiaojie0815, 2022-04-08:
@lukasmatena Since I don't understand the compilation character table. Please help submit these two characters in the character table, thank you very much.

bubnikv, 2022-04-08:
We are considering to process the dictionary and add the missing characters
into the OpenGL character texture on language initialization. That would
solve the issue for good.

xiaojie0815, 2022-04-12:
@lukasmatena Thank you very much, you worked hard