prusa3d/PrusaSlicer #8172: Ender 3S1

wamsuperman, 2022-04-07:
### Description of the bug

I am thinking that the Gcode gets messed up when you add a Z-lift I can get prints to work untill I add a z-lift it seems like this stops extruding after a certain point. I am able to get it to manualy extrude but prusa slicer messes up everytime

### Project file & How to reproduce

Ziped file is too big

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 11 and octoprint

### Printer model

Ender 3 s1

wamsuperman, 2022-04-07:
Was using the defualt profile on my ender 3 s1 it appears its the z lift when i add that to it it kills print after layer like 3

wamsuperman, 2022-04-07:
[GPS CAP v1_36m_0.20mm_205C_PLA_ENDER3S1(1).zip](https://github.com/prusa3d/PrusaSlicer/files/8445882/GPS.CAP.v1_36m_0.20mm_205C_PLA_ENDER3S1.1.zip)

rtyr, 2022-04-08:
I think there is no issue with the g-code. I printed it without issues as it is on MK3.

Maybe try it slice again without post-processing made by Octoprint plugin and print it directly from the printer. It may be some Octoprint related issue.

[image]

rtyr, 2022-04-13:
No response. The g-code is OK. Closing.

closed 2022-04-13 by rtyr

wamsuperman, 2022-04-14:
> Please provide some data. At least some example project file and that
> problematic g-code file, it shouldn't be that big. Are you using bundled
> profiles or your own profiles?

Just saw this it was in spam folder it only has a problem once I add zlift
I don't know why. I have stopped using it and have even been able to speed
up my prints using the alpha profile with modified points but as soon as I
ad a z lift it stops extruding and just prints in air.