prusa3d/PrusaSlicer #8173: SKIRT AND BRIM ERROR

NZD4X, 2022-04-07:
### Description of the bug

VERSION 2.4.1 + win64
SETTING AN INNER BRIM ONLY PRODUCES AN OUTER BRIM ONLY

### Project file & How to reproduce

![image](https://user-images.githubusercontent.com/46657967/162270270-89bdade3-12a1-4e7e-aee5-95a2
[image]
860c834c.png)
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

VERSION 2.4.1 + win64

### Operating system

Win 11

### Printer model

MK3 i3

rtyr, 2022-04-07:
Skirt and brim are two separate features. You have to set "Brim width" value, not number of skirt loops.

rtyr, 2022-04-07:
I believe it was answered, I am closing this issue.

closed 2022-04-07 by rtyr