prusa3d/PrusaSlicer #8174: Open zip files containing multiple STLs

j9brown, 2022-04-08:
**Is your feature request related to a problem? Please describe.**

I often design and print objects that consist of several parts using OnShape.  OnShape has a handy feature to export all of the parts as a ZIP file that contains multiple STLs.  I'd like to be able to open this ZIP file directly in PrusaSlicer without having to extract the STLs first.

**Describe the solution you'd like**

I'd like the Import Files dialog in PrusaSlicer to allow me to select a ZIP file to open.  When opened, PrusaSlicer would scan the archive for all models of types that it recognizes (STL, STEP, etc.) and add them to the Plater.

**Describe how it would work**

There are numerous open source libraries available for reading the contents of ZIP archives.  PrusaSlicer probably uses one already somewhere.  It would suffice to add some code to loop over all files of known model types within the imported ZIP archive, decompress them in memory (perhaps as a stream), and add the models to the Plater as usual.

**Describe alternatives you've considered**

Currently, I manually extract the files from the archive to a temporary directory and then import them into PrusaSlicer from that directory.  It works but it's a bit tedious to do when I'm iterating on a design that has many parts.

**Additional context**

Thanks!

j9brown, 2023-03-22:
Looks like this feature has been implemented in v2.6!  Yay!

lukasmatena, 2023-03-22:
Correct. Closing.

closed 2023-03-22 by lukasmatena