prusa3d/PrusaSlicer #8176: Filament usage estimates differ between UI and output file

DrLex0, 2022-04-08:
### Description of the bug

The total filament used estimate that is written to gcode files is not the same as what is shown in the UI. 

[image]
[image]

After some investigating, the difference is that the UI includes extrusions performed in the start G-code (shown as “Custom” in the viewer), while the output file does not. The “total filament used” and “total filament cost” values in the G-code file should also include the “custom” estimate, otherwise it is not really total.

### Project file & How to reproduce

[Project-and-output.zip](https://github.com/prusa3d/PrusaSlicer/files/8450866/Project-and-output.zip)

1. Slice and export
2. Compare Sliced Info in the UI with the comments in the gcode file

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.2-rc1+x64

### Operating system

Mac OS 10.14.6

### Printer model

FlashForge Creator Pro

foreachthing, 2022-04-08:
Seams to be fixed in 2.5.0 a0:
(it's Windows, though...)
[image]

DrLex0, 2022-04-08:
@foreachthing Try it with the .3mf file attached to this issue. If your start G-code doesn't do any extrusions, you won't encounter this issue.

foreachthing, 2022-04-08:
That extrudes G1 F100 E23... does it not?

DrLex0, 2022-04-08:
It looks like only G1 commands with an X or Y argument are treated as extrusions that consume filament. Your G1 command is also counted as an (un)retract, and is not added to the ‘custom’ count. Try adding X and/or Y movement, and you'll see it will be counted.

closed 2026-09-29 by prusaslicer-bot[bot]